gphoto/libgphoto2 #380: Set config not succeed but return GP_OK

xuyao178, 2019-02-28:
**Describe the bug**

I use the gp_camera_set_single_config function to set "copyright" and return GP_OK when camera busy. 
In fact, it did not succeed. The copyright in camera hasn't changed.

After I read the code，Return GP_OK is because the code in camlibs/ptp2/config.c as below:
if (mode == MODE_SINGLE_SET)
   return GP_OK;
Does this code have any special purpose?

But I can use  find_widget_by_name to get the same copyright that set at first（not the same in camera）.  I don't know why.

**Name the camera**
Canon EOS 100D

**libgphoto2 and gphoto2 version**
2.5.22

xuyao178, 2019-02-28:
I try to change the code in config.c as below:
```
if (mode == MODE_SINGLE_SET)
   return GP_OK;
```
to:
```
if (mode == MODE_SINGLE_SET)
   return ret;
```
I cant get the error code. But the config value have been cached in the Camera struct.
In my side, I can reinitialization the camera to resolve the problem. But maybe it not a good way.

msmeissn, 2019-03-03:
I applied this fix to git.
the configs are not really cached in the camera struct though. can you get a debug trace perhaps?

xuyao178, 2019-03-08:
The critical code as below when I call gp_camera_set_config.

camlibs/ptp2/ptp.c
```
uint16_t
ptp_canon_eos_setdevicepropvalue (PTPParams* params,
	uint16_t propcode, PTPPropertyValue *value, uint16_t datatype
) {
......
          switch (propcode) {
......
          case PTP_DTC_STR:
			strcpy((char*)data + 8, value->str);
			free (params->canon_props[i].dpd.CurrentValue.str);
			params->canon_props[i].dpd.CurrentValue.str = strdup(value->str); 
			break;
	}
......
        ret = ptp_transaction(params, &ptp, PTP_DP_SENDDATA, size, &data, NULL);
	free (data);
	return ret;
}
```
The config will store in "&camera->pl->params->canon_props[i]". however the ptp_transaction function return GP_OK or not.  
So, if  ptp_transaction return failed. I will get the wrong config value through gp_camera_get_config. Because gp_camera_get_config will read the value in "&camera->pl->params->canon_props[i]" directly.

msmeissn, 2019-03-09:
I split the switch code into two, and now commit the new values only after successful ptp call.

msmeissn, 2019-04-07:
it is solved I think?

closed 2019-04-07 by msmeissn