gphoto/libgphoto2 #382: Canon G7X developement: Capture works fine but could not capture image error is shown

SparkyCola, 2019-03-03:
When captureing (to SD card) gphoto gives me the following output:
```
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /store_00010001/DCIM/121___03/IMG_2384.JPG on the camera
```
Which I traced back to [this line](https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/library.c#L5103). I've verified it with a debug message and found out I get button=0 returned, also there are further errors in  the [log](https://github.com/gphoto/libgphoto2/files/2923364/my-logfile.txt). Anyway I was wondering what this means in order to know whether I can safely ignore it or should investigate further.

msmeissn, 2019-03-03:
can you try if it works better over USB instead of PTPIP?

SparkyCola, 2019-03-03:
I tried USB but the camera is not automatically detected (lsusb lists it but that's it, the camera also only responds to the playback button so I'm not sure what's going on) and I still have to figure out how to get into PTP-mode, I could use CHDK but that wouldn't be the same. I don't now if I expressed this clear enough above, the issue is only that it shows an error which doesn't seem to exist.