gphoto/libgphoto2 #384: --capture-image-and-download error -6 with EOS M100

nadre, 2019-03-04:
Hi there, awesome project!

When running 

`
gphoto2 --port usb: --capture-image-and-download
`

on my Raspy connected to a EOS M100, I'm getting this error:

`
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') *** 
`

Any ideas?

Here's the log:

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/2928069/my-logfile.txt)

Best, Erdan.

xuyao178, 2019-03-08:
It's seems your libgphoto2(2.5.12) version too old. Try to upgrade it to newest version.

msmeissn, 2019-03-09:
yes, the libgphoto2 version needs to be updated for EOS M support.

closed 2019-03-09 by msmeissn

nadre, 2019-03-11:
yes, it worked. thanks.

Heinnn, 2019-04-27:
how to update the libgphoto2?
I tried to download from https://sourceforge.net/projects/gphoto/files/
then 
autoreconf --install --symlink
./configure
make
sudo make install

but when I typed gphoto2 --version, It still be a libgphotos 2.5.12
Anyway to update???

Thanks