gphoto/libgphoto2 #385: install from github camera not detected

pdeman, 2019-03-07:
Hi
I tried to install libgphoto2 from github without success.
I followed what is described [here](https://github.com/gphoto/libgphoto2/blob/master/INSTALL) .

it generated this:
`/usr/local/bin$ ls`

> gphoto2-config  gphoto2-port-config
(shouldn't I have gphoto2 as well ?)

and 
`:/usr/local/lib$ ls *gphoto*
`
> libgphoto2.la       libgphoto2_port.so     libgphoto2_port.so.12.0.0  libgphoto2.so.6
> libgphoto2_port.la  libgphoto2_port.so.12  libgphoto2.so              libgphoto2.so.6.1.0
> 
> libgphoto2:
> 2.5.22.1  print-camera-list
> 
> libgphoto2_port:
> 0.12.0

at the end I get errors 
` ./sample-autodetect `

> *** Contexterror ***              
> Could not detect any camera
> No camera auto detected.

nvidia@tegra-ubuntu:~/Desktop/Canon
I have a Canon EOS 5DSR but it doesn't see it.

xuyao178, 2019-03-08:
libusb installed in your linux?

pdeman, 2019-03-08:
yes I have:
checking for libusb to use... autodetect
checking for LIBUSB... yes
checking usb.h usability... yes
checking usb.h presence... yes
checking for usb.h... yes
checking for function usb_open in libusb... yes
checking libusb library flags... "-lusb"
checking libusb cpp flags... ""
 when doing the `./configure --prefix=/usr/local`

and at the end of the configure:

> Configuration (libgphoto2 2.5.22.1):
> 
>   Build
>     Source code location:          .
>     GP2DDB support:                no  (EXPERIMENTAL)
>     Compiler:                      gcc
>     libltdl includes:              
>     libltdl library:               -lltdl
> 
>   Features
>     Use translations:              yes
>     JPEG mangling support:         no
>     libxml-2.0:                    yes
>     XML support for Olympus E s..: yes
>     gdlib:                         no
>     libGD conversion support:      no
>     Camlibs:                       all
>     build API docs with gtk-doc:   no (not requested)
>     libexif:                       no
> 
>     Subprojects:                   libgphoto2_port

pdeman, 2019-03-08:
ok updating libusb-dev and some other dependencies solved the problem.
./sample-autodetect worked.

but I don't have : "gphoto2 the command line (CLI) interface to libgphoto2."
how do I get it ?

msmeissn, 2019-03-09:
The gphoto2 commandline tool is an additional repo which needs to be seperately built, see github.com/gphoto/gphoto2

pdeman, 2019-03-14:
ok but when I compile gphoto2 it links to the libgphoto2 version from the apt-get and not the one that I compiled using this git. https://github.com/gphoto/gphoto2/issues/206