gphoto/libgphoto2 #386: Olympus OM-D E-M5 mark II not working anymore

berylius, 2019-03-09:
I can't get the camera to work with libgphoto2 in PC mode.
When i use MTP mode it seems to work but then of course i lack the capture functionnality.

Camera has Firmware 4.1
System is running Ubuntu mate 18.04.

Here are the results of different commands I tried:
gphoto2 --auto-detect 
> Olympus E-M1                   usb:001,009     

gphoto2 --version
> gphoto2         2.5.20         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
> libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
> libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

gphoto --summary
[debug.log](https://github.com/gphoto/libgphoto2/files/2948254/debug.log)

If any body can help. I can try to capture USB trafic if neccessary (just need some explanation for that ;) )

msmeissn, 2019-03-09:
could you try building with libusb1 ? we might need the async sent interrupt handled correctly.

berylius, 2019-03-09:
I don't how to do that :p !

For information, I have tested it on a another computer through the software Kstars with Indi (https://github.com/indilib/indi) drivers and it works. 

I just needed to install gphoto2, i got it from ubuntu repository so it's only version 2.5.15 but it worked.
Here is the log file. And as you proposed, the libusb1 was installed on this computer.

[debug (3).log](https://github.com/gphoto/libgphoto2/files/2948388/debug.3.log)

closed 2019-03-12 by berylius

cparg, 2019-08-03:
I have the same problem.
I updated libusb to the latest version.
Now, the first 2 preview-captures work and the third and all following fail.
here are the debug log files for the first three runs:

- [debug1.txt.gz](https://github.com/gphoto/libgphoto2/files/3464366/debug1.txt.gz)
- [debug2.txt.gz](https://github.com/gphoto/libgphoto2/files/3464367/debug2.txt.gz)
- [debug3.txt.gz](https://github.com/gphoto/libgphoto2/files/3464368/debug3.txt.gz)