gphoto/libgphoto2 #392: Problem Using Memory Card Canon SX740 HS

chalmersb, 2019-03-16:
When using gphoto2 with a Canon PowerShot SX740 HS with a memory card in, it is only capable of executing one command that manipulates the camera. The second command always fails. The camera has an error pop up on the screen that says: "Err 70 An error prevented shooting. Turn the camera off and on again or re-install the battery." 

For every command, the camera retracts the lens and extends the lens again. Any command that interacts with the camera, including commands like --trigger-capture, --capture-image-and-download, and --get-config, all cause the camera to retract and extend the lens. If something like --capture-image-and-download -I 3 -F 5 is run, it will work, and the lens will retract and extend only for the first image but will remain extended to take the rest of the images. However, the next command entered that causes the lens to retract and extend will fail.  --capture-tethered works in the same way in that it will take pictures until you stop it the first time but will fail the second time. Commands like --auto-detect and --set-config do not cause this to happen and can be run repeatedly without fail. 

To prevent this error, the camera has to either be turned off and on again in between each command, or the usb cord has to be unplugged and plugged in again. I attempted to use --reset in between commands, but that did not work. I also attempted the following command I found on a thread on source forge, but that also did not work: 
gphoto2 --wait-event=1s --set-config eosremoterelease="Press 1" --wait-event=1s --set-config eosremoterelease="Press 2" --wait-event=100ms --set-config eosremoterelease="Release 2" --set-config eosremoterelease="Release 1" --wait-event-and-download=5s

The error that is displayed in the terminal after failure is
*** Canon EOS M Full-Press failed (0x2019: PTP Device Busy) ***
Error: Could not capture image.
Error: Could not capture.
*** Error (-110: 'I/O in progress) ***

I removed gvfs and killed any remaining gvfs processes, so that is not the cause of the issue. I’ve also tried using shell mode, but the problem persists.

When a memory card is in the camera, the error always occurs with the second command when capturetarget is set to card. When capturetarget is set to sdram, it is slightly less unreliable but still fails within four commands that cause the lens to retract and extend. However, when there is no memory card in the camera, this error rarely occurs. I’ve run a time lapse a few times with 400 frames and 3 second intervals, and this error only happened during one time lapse after 280 photos.

I'm not sure if this is important, but I put a micro SD card adapter with nothing in it in the camera to see what would happen, and the camera recognized an SD card was present. A message appeared on the screen of the camera that says the card cannot be accessed, and gphoto2 still only works the first time with this setup.

When --auto-detect is run, the camera is named as "Canon PowerShot SX740 HS". Here is the output of gphoto2 –version: 
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

The computer I am using with gphoto2 is a Raspberry Pi 2 Model B V1.1 with Raspbian Stretch.

I have log files for --capture-image-and-download once and then a second time, for both capturetarget set to card and sdram. The files for each of the first instances are too large, so I'm not sure how to attach them.

elstoc, 2019-10-09:
Similar issue with 760D - errors only when an SD card is present. https://github.com/gphoto/gphoto2/issues/131

dlwalter, 2019-11-06:
Having the same problem with the SX740HS

```
gphoto2 --auto-detect --capture-image  --set-config capturetarget=1
Model                          Port
----------------------------------------------------------
Canon PowerShot SX740 HS       usb:001,021
New file is in location /capt0000.jpg on the camera

*** Error ***
Failed to set new configuration value 1 for configuration entry capturetarget.
*** Error (-110: 'I/O in progress') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --auto-detect --capture-image --set-config capturetarget=1

Please make sure there is sufficient quoting around the arguments.
```

Camera screen says "Err 70 An error prevented shooting. Turn the camera off and on again or re-install the battery