gphoto/libgphoto2 #395: interval between two capture extremely long, canon eos 5DS R

pdeman, 2019-03-19:
Hi, 
I am using a canon 5DS R and 
gphoto2         2.5.20.1       gcc, popt(m), no exif, cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22.1       all camlibs, gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

I want to trigger image captures as quickly as possible. Images must be RAW and saved on the SD card on the canon, I don't need to download them on the computer.

I took as example simple-trigger-capture and have this code:
[sample-trigger-capture.txt](https://github.com/gphoto/libgphoto2/files/2983344/sample-trigger-capture.txt)

the time between two images is 1.8secondes ...
I need to be at at least one image seconde.

here is the result of the command :` gphoto2 --capture-image -F 10 -I 1 --debug --debug-logfile=debug.log`
which is probably the closest to the c code.
[debug.log](https://github.com/gphoto/libgphoto2/files/2983377/debug.log)

the SD card is supposed to be 160 Mb/s writting, so the writing of an image should take less than 0.5s.

pdeman, 2019-03-25:
in library.c I see many : GP_LOG_D
where does it saves the log ?
is it always enabled ? 
does it work when not using through gphoto2 CLI but with my sample-trigger-capture.c ?

msmeissn, 2019-03-27:
the sample trigger capture program logging can be adjusted here:
        gp_log_add_func(GP_LOG_ERROR, errordumper, NULL);
        /*gp_log_add_func(GP_LOG_DATA, errordumper, NULL); */
if you switch it to LOG_DATA, it will log everythign.

pdeman, 2019-03-27:
ok thanks. in case it can help, here is my code and associated logfile.
[test.txt](https://github.com/gphoto/libgphoto2/files/3012752/test.txt)
[triggerCustomized.txt](https://github.com/gphoto/libgphoto2/files/3012753/triggerCustomized.txt)
I don't get why there is as many ptp request/response all the time when I just want to trigger a capture "no matter what" if the canon is ready of course, but i don't care about the focus, etc ... I want it to trigger. I am in embedded configuration, if it doesn't trigger at the good timing, data is lost, I prefer bad quality image than lost data.