gphoto/libgphoto2 #398: Read .mjpeg (and stop?)

aolbrechts, 2019-03-25:
Hello,

I'm writing a small page/app that is used via the web to control the basic functions of a camera (change settings + take picture and download). 

It rather works as I want for the moment (haven't implemented the timelapse functions I'm planning on) but I have issues with the liveview capture.

I'm using this code to start the capture : 
exec("gphoto2 --capture-movie --stdout | ffmpeg -i pipe:0 /var/www/html/fifo.mjpg");

I get a fifo.mjpg file but I can't open it anywhere (even VLC ... ?). I can't display it in a webpage either, I only get the first frame.

What could I do to verify that the live capture works?

(additional question : elsewhere in my script I have to end the live capture but I can't find what code to use? I have the code to bring the mirror down but it doesn't work because the live capture process is still in action).

Any help on this? Thanks for your help!

pReya, 2019-10-18:
@aolbrechts Did you find a solution for this, yet?

aolbrechts, 2019-10-18:
Unfortunately, no :(

[Adret & Ubac]<https://www.adret-ubac.be>       Antoine Olbrechts   |   Web & Social media Consultant   |   Adret & Ubac
[http://www.carmeldesign.be/signature_email/email_16.png]       antoine@adret-ubac.be<mailto:antoine@olbrechts.eu>              [http://www.carmeldesign.be/signature_email/web_16.png]         www.adret-ubac.be<https://www.adret-ubac.be>            [http://www.carmeldesign.be/signature_email/twitter_16.png]     @aolbrechts<https://twitter.com/aolbrechts>             [http://www.carmeldesign.be/signature_email/facebook_16.png]    sur facebook<http://www.facebook.com/antoine.olbrechts.eu>
5 sur les heids | 4053 Embourg | Belgique
Tél +32 496 879 865

msmeissn, 2019-10-20:
to verify it works, you can run gphoto2 --capture-movie=10s  on the commandline and see if that is playable in e.g. mplayer?

Parallel instances of gphoto2 accessing the same device will be tricky, as the cameras only support 1 session, so if you want to do configuration, capture and preview capture in parallel, it needs to be in one process.

Someone has likely written something like it already, i dont know it from the top of my head though.

pReya, 2019-10-21:
To look at the live preview in VLC, you don't need ffmpeg or a FIFO, just the right CLI parameter for VLC. You need to specifiy a framerate at which the MJPEG should be played. You can test the framerate of your cam if you just run `gphoto2 --capture-movie=10s`. After 10s you will see how many frames were captured. Just divide this by 10 and you have a good value to put into this VLC parameter: `gphoto2 --capture-movie --stdout | vlc - --mjpeg-fps 30`