gphoto/libgphoto2 #399: Compile options when calling directly function from library.c

pdeman, 2019-03-26:
Hi,
I am trying to call directly some functions from the library.c file such as camera_trigger_canon_eos_capture
but I can't find the good gcc configuration.

here is my code:

[triggerCustomized.txt](https://github.com/gphoto/libgphoto2/files/3009036/triggerCustomized.txt)

and my gcc command:
`gcc -O2 -Wall -I../libgphoto2/ -I../libgphoto2_port/gphoto2/ -I../libgphoto2/libgphoto2/ -I/usr/include/libxml2/ -Wl,--start-group -L/usr/local/lib -lgphoto2 -lgphoto2_port -L/lib/aarch64-linux-gnu/ -lltdl -Wl,--end-group -o triggerCustomized triggerCustomized.c context.c config.c ../camlibs/ptp2/library.c  ../camlibs/ptp2/ptp.c ../camlibs/ptp2/usb.c ../libgphoto2/gphoto2-context.c ../libgphoto2/gphoto2-filesys.c`

but I get the following error, how can I solve this ?:

> triggerCustomized.c:(.text.startup+0x48): undefined reference to `gp_log_add_func'
> triggerCustomized.c:(.text.startup+0x50): undefined reference to `gp_camera_new'
> triggerCustomized.c:(.text.startup+0x5c): undefined reference to `gp_camera_init'
> triggerCustomized.c:(.text.startup+0xa4): undefined reference to `camera_trigger_canon_eos_capture'
> triggerCustomized.c:(.text.startup+0x13c): undefined reference to `gp_camera_wait_for_event'
> triggerCustomized.c:(.text.startup+0x214): undefined reference to `gp_camera_exit'
> /tmp/ccaL91ck.o: In function `get_config_value_string':
> config.c:(.text+0x48): undefined reference to `gp_camera_get_single_config'
> config.c:(.text+0x60): undefined reference to `gp_widget_get_type'
> config.c:(.text+0x90): undefined reference to `gp_widget_get_value'
> config.c:(.text+0xac): undefined reference to `gp_widget_free'
> config.c:(.text+0x10c): undefined reference to `gp_camera_get_config'
> config.c:(.text+0x128): undefined reference to `gp_widget_get_child_by_name'
> config.c:(.text+0x138): undefined reference to `gp_widget_get_type'
> config.c:(.text+0x1e4): undefined reference to `gp_widget_get_child_by_label'
> /tmp/ccaL91ck.o: In function `set_config_value_string':
> config.c:(.text+0x25c): undefined reference to `gp_camera_get_config'
> config.c:(.text+0x274): undefined reference to `gp_widget_get_child_by_name'
> config.c:(.text+0x284): undefined reference to `gp_widget_get_type'
> config.c:(.text+0x2cc): undefined reference to `gp_widget_free'
> config.c:(.text+0x308): undefined reference to `gp_widget_set_value'
> config.c:(.text+0x324): undefined reference to `gp_camera_set_single_config'
> config.c:(.text+0x33c): undefined reference to `gp_camera_set_config'
> config.c:(.text+0x3c4): undefined reference to `gp_widget_get_child_by_label'
> /tmp/ccaL91ck.o: In function `canon_enable_capture':
> config.c:(.text+0x468): undefined reference to `gp_camera_get_single_config'
> config.c:(.text+0x478): undefined reference to `gp_widget_get_type'
> config.c:(.text+0x4b0): undefined reference to `gp_widget_free'
> config.c:(.text+0x4e8): undefined reference to `gp_widget_set_value'
> config.c:(.text+0x504): undefined reference to `gp_camera_set_single_config'
> /tmp/ccaulv2Y.o: In function `print_debug_deviceinfo':
> library.c:(.text+0x2c): undefined reference to `GP_LOG_D'
> library.c:(.text+0x3c): undefined reference to `GP_LOG_D'
> library.c:(.text+0x4c): undefined reference to `GP_LOG_D'
> library.c:(.text+0x5c): undefined reference to `GP_LOG_D'
> library.c:(.text+0x6c): undefined reference to `GP_LOG_D'
> /tmp/ccaulv2Y.o:library.c:(.text+0x7c): more undefined references to `GP_LOG_D' follow
> /tmp/ccaulv2Y.o: In function `gpfile_putfunc':
> library.c:(.text+0x1f0): undefined reference to `gp_file_append'
> /tmp/ccaulv2Y.o: In function `gpfile_getfunc':
> library.c:(.text+0x240): undefined reference to `gp_file_slurp'
> /tmp/ccaulv2Y.o: In function `ptp_debug_func':
> library.c:(.text+0x5f8): undefined reference to `gp_logv'
> /tmp/ccaulv2Y.o: In function `strcpy_mime':
> library.c:(.text+0x828): undefined reference to `GP_LOG_D'
> /tmp/ccaulv2Y.o: In function `debug_objectinfo.isra.5':
> library.c:(.text+0x870): undefined reference to `GP_LOG_D'
> library.c:(.text+0x880): undefined reference to `GP_LOG_D'
> library.c:(.text+0x890): undefined reference to `GP_LOG_D'
> library.c:(.text+0x8a0): undefined reference to `GP_LOG_D'
> /tmp/ccaulv2Y.o:library.c:(.text+0x8b0): more undefined references to `GP_LOG_D' follow
> /tmp/ccaulv2Y.o: In function `set_mimetype':
> library.c:(.text+0xb78): undefined reference to `gp_file_set_mime_type'
> library.c:(.text+0xb98): undefined reference to `GP_LOG_D'
> library.c:(.text+0xbb0): undefined reference to `gp_file_set_mime_type'
> /tmp/ccaulv2Y.o: In function `folder_to_handle.constprop.24':
> library.c:(.text+0xc4c): undefined reference to `GP_LOG_D'
[... 999 lines cut ...]
> gphoto2-filesys.c:(.text+0x3014): undefined reference to `gp_file_unref'
> gphoto2-filesys.c:(.text+0x3024): undefined reference to `gp_file_ref'
> gphoto2-filesys.c:(.text+0x3038): undefined reference to `gp_file_unref'
> gphoto2-filesys.c:(.text+0x3048): undefined reference to `gp_file_ref'
> gphoto2-filesys.c:(.text+0x305c): undefined reference to `gp_file_unref'
> gphoto2-filesys.c:(.text+0x306c): undefined reference to `gp_file_ref'
> gphoto2-filesys.c:(.text+0x3080): undefined reference to `gp_file_unref'
> gphoto2-filesys.c:(.text+0x3090): undefined reference to `gp_file_ref'
> gphoto2-filesys.c:(.text+0x30a4): undefined reference to `gp_file_unref'
> gphoto2-filesys.c:(.text+0x30b4): undefined reference to `gp_file_ref'
> gphoto2-filesys.c:(.text+0x30d4): undefined reference to `GP_LOG_D'
> gphoto2-filesys.c:(.text+0x3124): undefined reference to `gp_file_get_data_and_size'
> gphoto2-filesys.c:(.text+0x3174): undefined reference to `gp_file_get_data_and_size'
> gphoto2-filesys.c:(.text+0x319c): undefined reference to `GP_LOG_D'
> gphoto2-filesys.c:(.text+0x31ac): undefined reference to `GP_LOG_D'
> gphoto2-filesys.c:(.text+0x3234): undefined reference to `gp_setting_get'
> gphoto2-filesys.c:(.text+0x3278): undefined reference to `GP_LOG_D'
> gphoto2-filesys.c:(.text+0x32b0): undefined reference to `gp_setting_set'
> /tmp/ccbXk5Dx.o: In function `gp_filesystem_set_info_noop':
> gphoto2-filesys.c:(.text+0x3320): undefined reference to `C_PARAMS'
> /tmp/ccbXk5Dx.o: In function `gp_filesystem_set_info':
> gphoto2-filesys.c:(.text+0x3434): undefined reference to `C_PARAMS'
> /tmp/ccbXk5Dx.o: In function `gp_filesystem_get_storageinfo':
> gphoto2-filesys.c:(.text+0x35f8): undefined reference to `C_PARAMS'
> collect2: error: ld returned 1 exit status

msmeissn, 2019-03-26:
gp_camera_trigger_capture will trigger capture (without waiting for the image to be capture)... you need to poll the events

gp_camera_capture will capture the image and wait for the image to be added and returns it.

What is missing in the sample-trigger-capture.c example?

pdeman, 2019-03-26:
it s taking forever between two trigger. sometimes 3 secondes. i need to shorter this. i am trying to understand what can cause this.

msmeissn, 2019-03-26:
in that case trying to log debug output might help... 

there might be focusing happening? there is a 2 second max time for focusng.

Is this latest 2.5.22 libgphoto2 code?

pdeman, 2019-03-26:
no it s not the focussing. it s the multiple wait event which are taking forever.

msmeissn, 2019-03-26:
a debug logfile would help... the log functions are not commented it, but you can try the commandline tool

gphoto2 --trigger-capture --trigger-capture --wait-event-and-download=20s --debug --debug-logfile=xx.log

msmeissn, 2019-03-26:
i saw your other bug where you provided this ... will check this out

pdeman, 2019-03-27:
I'ld still like times to times to call the lowest possible function to trigger the camera.
every ms won is good to me.
I'ld like to do the camera configuration using the libgphoto api as shown in documentation and in example.
but to trigger photos, I know I am using a canon eos 5dsr, that I capture photo on the compact flash on the device and won't tranfer it on my computer, I'll just need the file name, I'ld like to call directly the good trigger function. if it can make me win in total 100 ms between two shots I take it.

how can I configure my gcc command to be able to call the function in library.c ?

pdeman, 2019-03-29:
ok I added some include and modified the makefile.in in /examples to be able to use directly the function from library.c. it seems to work.

msmeissn, 2019-05-05:
seems solved I think

closed 2019-05-05 by msmeissn