gphoto/libgphoto2 #401: Possible upcoming support for the Rx0 II ?

jwoodrow, 2019-03-28:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

DSC-Rx0 II

**USB IDs**

**camera summary output**

**camera configuration output**

**test capture**

----------------------------------------------------------------------------------------

Hi, I've been working with gphoto2 and an Rx0 (first generation) for a couple of months now and hit a physical limitation for the framerate I can effectively get when using live view (max ~24fps at 1024x680 and I believe the limitation is from the camera only supporting USB 2.0).

Anyway, I saw that the Rx0 mark II was coming out this April and was wondering if:
- What are the chances it would simply work out of the box with gphoto2 if I were to buy one to replace the mark I in my projet ?
- If it isn't compatible how long would you think it would take to be able to make it compatible ?

I know this isn't really a new camera request per-se, since I can't really get my hands on an Rx0 II yet. But since sometimes this kind of information flies under the radar I thought I could try and anticipate the need for this camera to be supported.

Once it comes out and I get my hands on one I'll keep the issue updated with the information I could not fill out yet, as well as check if it isn't already compatible out of the box so it could then be added to the supported camera list.

msmeissn, 2019-03-28:
Sony? Usually the new ones work ... But I cannot guarantee it

jwoodrow, 2019-03-30:
Thanks for the reply, once it comes out and I manage to get my hands on one I'll let you know if it works fine out of the box or not so you can then decide if you wish to add it to the list of supported cameras after 👍

Radiofreak1041, 2019-09-23:
Hi! Do you know if it's working with the DSC-RX0 ii yet?

msmeissn, 2019-09-29:
was reported working in issue 419, commit f4618dfcb0bc1c45accb311cf2931502c3e73eba  adds thje id ... will be in the next release (but will already work even with the current release)

closed 2019-09-29 by msmeissn