gphoto/libgphoto2 #402: Sony A7R III - Step Focus

andyoest, 2019-04-01:
**Is your feature request related to a problem? Please describe.**

no

**Describe the solution you'd like**

I'm looking for a possibility to control the focus position of the Sony A7R III via Terminal.
Obtaining pixel shift images via the gphoto2 CLT works perfectly.

**Describe alternatives you've considered**

Automating Sony Remote App is not possible out of the box. There are products from Camfy allowing Focus Stacking but this would just introduce new problems into our worklfow.

**Additional context**
We're working on a solution for high resolution photogrammetry applications. We're trying to combine Sony's pixel shift with focus stacking to obtain the best image possible.

We do not have any experience reverse engineering the USB calls necessary but alot in computer vision and 3D reconstruction. We're happy to share our scripts and code as soon as we have the everything up and running.

Best,
Andy

Hutch67, 2019-10-16:
@andyoest see issue #434 for a solution
https://github.com/gphoto/libgphoto2/issues/434

msmeissn, 2019-10-19:
if this is manual focus drive, then this is possible , including the bugfix referenced in issue 434