gphoto/libgphoto2 #404: gphoto2 --list-files is slowly on Nikon camera

xuyao178, 2019-04-04:
I use gphoto2 --list-files to show the files in camera.  It's cost 38.2s to list 766 files.  Is it the reason of camera itself？

xuyao178, 2019-04-04:
#754   _DSC8412.NEF               rd 35751 KB application/x-unknown
#755   _DSC8413.JPG               rd  4089 KB 5520x3680 image/jpeg
#756   _DSC8413.NEF               rd 36282 KB application/x-unknown
#757   _DSC8414.JPG               rd  4096 KB 5520x3680 image/jpeg
#758   _DSC8414.NEF               rd 36287 KB application/x-unknown
#759   _DSC8415.JPG               rd  4094 KB 5520x3680 image/jpeg
#760   _DSC8415.NEF               rd 36284 KB application/x-unknown
#761   _DSC8416.JPG               rd  4071 KB 5520x3680 image/jpeg
#762   _DSC8416.NEF               rd 36256 KB application/x-unknown
#763   _DSC8417.JPG               rd  4102 KB 5520x3680 image/jpeg
#764   _DSC8417.NEF               rd 36291 KB application/x-unknown
#765   _DSC8418.JPG               rd  4138 KB 5520x3680 image/jpeg
#766   _DSC8418.NEF               rd 36299 KB application/x-unknown

real	0m38.480s
user	0m0.364s
sys	0m0.337s

msmeissn, 2019-04-07:
It is likely the camera itself taking so long.

gphoto2 -L --debug --debug-logfile=list-files.log

will generate a list-files.log with timestamps to check where it takes so much time