gphoto/libgphoto2 #407: Support for Sony Alpha-A6400

xmizrx, 2019-04-22:
I am planning on buying a Sony Alpha-A6400.
Could you please support it.

msmeissn, 2019-05-05:
We have so far not received any reports for it, but it likely works already as-is.

solexx, 2019-08-18:
I own an a6400 and use it with [sony-pm-alt ](https://github.com/falk0069/sony-pm-alt), just like I did previously with my a6000. There is only one annoyance: the a6400 does not automatically shut down after sending the last image. For some reason the camera keeps "sending" without actually transmitting anything. I can provide a debug log if you think that might be helpful. But I have to admit that I am not sure whether this is actually an issue with gphoto or sony-pm-alt.

pReya, 2019-10-21:
Support for the a6400 was just added. So this Issue can be closed. https://github.com/gphoto/libgphoto2/commit/f6042525c5dec3f58b1439e03547e7e7f157e32c

closed 2019-10-21 by msmeissn