gphoto/libgphoto2 #408: Virtual GPhoto device driver for testing client applications

yecril71pl, 2019-04-22:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*
Virtual GPhoto

**USB IDs**
TBD

**camera summary output**
TBD

**camera configuration output**
TBD

**test capture**
A predefined image

msmeissn, 2019-04-23:
libgphoto2 has a virtual PTP camera, configure with --enable-vusb  and no other libusb libraries.
a bit tricky to handle, but should work.

yecril71pl, 2019-04-25:
I cannot find it documented.  Not in INSTALL and not in [configure help](https://build.opensuse.org/package/live_build_log/home:yecril71pl:branches:graphics/libgphoto2-virtual/openSUSE_Leap_15.0/x86_64).

tkircher, 2023-03-16:
I'll add the documentation here since it doesn't seem to exist anywhere else.

In order for the virtual camera device to work, like @msmeissn suggested above, you must enable vusb and explicitly disable all other usb libraries:

`./configure --enable-vusb --without-libusb --with-libusb-1.0=no`

It is also important to make sure you don't have any other versions of the library installed. So, on a clean system without libgphoto2 or gphoto2 installed, build libgphoto2, then install it, and build gphoto2 and install it. The device doesn't come up as `vusb:`, rather
```
$ gphoto2 --list-ports
...
usb:001,001                      Universal Serial Bus
```

```
$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D750                 usb:001,001     
```

```
$ gphoto2 -a
Abilities for camera             : Nikon DSC D750                              
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Image
                                 : Preview
                                 : Trigger Capture
Configuration support            : yes
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : no
```

pmnlla, 2023-03-24:
This is done with the configuration script at build time, right?

yecril71pl, 2023-04-20:
I have [reconfigured the library](https://build.opensuse.org/build/home:yecril71pl:branches:graphics/openSUSE_Tumbleweed/x86_64/libgphoto2/_log) and it [did not work](https://build.opensuse.org/package/live_build_log/home:yecril71pl:branches:graphics/gphoto2-vusb-test/openSUSE_Tumbleweed/x86_64) :disappointed: 

> [   52s] + gphoto2 -a
> [   52s] Could not detect any camera

tkircher, 2023-04-20:
> I have [reconfigured the library](https://build.opensuse.org/build/home:yecril71pl:branches:graphics/openSUSE_Tumbleweed/x86_64/libgphoto2/_log) and it [did not work](https://build.opensuse.org/package/live_build_log/home:yecril71pl:branches:graphics/gphoto2-vusb-test/openSUSE_Tumbleweed/x86_64) 😞
> 
> > [   52s] + gphoto2 -a
> > [   52s] Could not detect any camera

Looks like you installed libgphoto2 from some package source, rather than building it:

> [   42s] libgphoto2_port12-2.5.30-250.1        ########################################

yecril71pl, 2023-04-20:
> Looks like you installed libgphoto2 from some package source, rather than building it:
> 
> > [   42s] libgphoto2_port12-2.5.30-250.1        ########################################

I have [packaged it](https://build.opensuse.org/build/home:yecril71pl:branches:graphics/openSUSE_Tumbleweed/x86_64/libgphoto2/_log) myself.

msmeissn, 2023-04-22:
i sent you a fix, osc rq accept -m ok 1082175

(basically FUZZ_PTP was undefined, so no camera was reported ... #define FUZZ_PTP makes it report the D750 id)

closed 2023-04-22 by yecril71pl