gphoto/libgphoto2 #409: Lumix HTTP/Wifi protocol

totoantibes, 2019-04-23:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*
All Lumix G, GX, GH series. (I can test on G80 and GH4)
seems that only GH4, GX80 and GH5 have a usb thethered mode available. Many other Lumix do not offer PTP tethered camera capture/liveview etc. Only option is to use an HTTP protocol over Wifi. (or maybe BLE for newer cams) there are many commands available via this protocol such as shutter speed, iso, aperture, quality, focus, zoom. rec start/stop, capture and capture cancel. also the images are avail to download to in RAW, JPG and Thumbs. A VGA/QVGA live view data stream is also available. 

There are various running examples of programs that do take advantage of this protocol  and woudl be great to open Gphoto2/ligphoto2 to Lumix cameras. Unfortunately the HTTP/Wifi connectivity is a bit of an issue to plug into the camlib/port structure of libgphoto as i understand it from a very high level.

here is a list of github availalble programs that take advantage of this protocol
-https://github.com/palmdalian/python_lumix_control 
- https://github.com/peci1/lumix-link-desktop (full app in java)
-https://github.com/Ultrawipf/qtpanaremote (simple QT based app in C)
-https://github.com/totoantibes/LumixCameraAscomDriver (my own Ascom driver for astrophotography written in VB.NET) it does implement the picture download of the last capture over http/get with retry after parsing the XML of the available pics. 
  
Use the Marketing name.

**USB IDs**
e.g. by running:   lsusb 

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

I am  a keen astrophotographer and would really like to use the libgphoto lib to interface with astrophoto session SW like ekos/kstars to control the Lumix CAMs. 

If you have no time/priority to dev this new family of cameras I woudl be keen to do it myself but some quidance/support would be needed prior to hacking something in this complex body of work

msmeissn, 2019-05-05:
This is something interesting to pursue :/ 
it needs a new camlib, perhaps a new port driver (or reuse ptpip somehow)

totoantibes, 2019-05-05:
well,

as a matter of  fact i started creating a new camlib (called lumix.c) and attached here. however 
1) i did only work in text editor. I dont have a dev env setup for gcc and linux. this is a little more than psuedo code. I have not even tried to compile it  once.
2) i am very rusty in C programming (20 years since my last project)  
3)  looked at libcurl and libxml for the meaty sections. with lots of picked up code from tutorial left an right
4) i tried to put comments where needed and included sample XML files that are needed to be parsed.
5) i started from an existing camlib (olympus sierra) . the config get  is still there and i tried to put in the usefull dictionary for the shutterspeed. (starting at line 365.
6) i dont know how i would get some info into this camlib such as the cameraIP address, shutterspeed, etc. so they are set as global variable on top,
7) not sure either how glibphoto handles the liveview feed. is it via camera_capture_ preview?  there is a "ability list" called GP_FILE_OPERATION_PREVIEW or GP_OPERATION_CAPTURE_PREVIEW but i did not find a method that actual makes use of that flag. however the lumix cam does open a UDP socket on a specified port and after some header bytes the video is really just an mjpeg in VGA mode. i did put also starting point to handle that but not sure if this is the right place in the camlib...
8) getting the captured image from the camera over HTTP/GET over wifi is a slow affair (10 sec'ish) varrying if the file is RAW of jpg or thumb. it also often fails and needs to be restarted, I did put in also some function in there to retstart DL from last known point 
9) so the code needs a non-blocking thread to take captures in bulb mode. intent is to have bulb shots of 300s and this cannot keep the driver blocked...

Basically as you can guess, i am full of good will here but need guidance

thanks

robert
[lumix.txt](https://github.com/gphoto/libgphoto2/files/3145715/lumix.txt)

totoantibes, 2019-05-24:
Markus,

I would really appreciate any help here. Thanks

msmeissn, 2019-05-26:
I tried inmtegrating your code into libgphoto2, but its already too big ... we should start very small from temlate, and get one initial command send first.
I think I need to rent a GH4 again myself :/

Sorry that I have not much time currentky

msmeissn, 2019-05-26:
ok, so I added a bui8lding version

it will take gphoto2 --port tcp:192.168.2.1 --camera "Panasonic LumixGSeries" --summary

msmeissn, 2019-05-26:
it is in the "lumix" branch , git checkout lumix   , autoreconf -i -f  ; ./configure ....  ; make

msmeissn, 2019-05-26:
I think not much more than camera_init() will be called now.

totoantibes, 2019-05-27:
Thanks Marcus,
I will investigate over the coming days,

totoantibes, 2019-06-05:
Hi Marcus, I was able to compile/debug the first stage and able to init and capture. How can i share my commits to the lumix branch? I am fairly new to github so not sure about the permissions/mechanics. 
[lumix.txt](https://github.com/gphoto/libgphoto2/files/3258183/lumix.txt)

msmeissn, 2019-06-07:
i committed this lumix.c to the lumix branch under your author name.

I am also still trying to organize both time and a Lumix vcamera ;) you know if a DC-TZ91 will be usable?

totoantibes, 2019-06-08:
thanks. I tried to find the list of Panasonic cameras supported by their image app.. but not found. however since it has wi-fi it is  a good indicator that it will work. I also found on the [google play  image app ](https://play.google.com/store/apps/details?id=com.panasonic.avc.cng.imageapp&hl=en)some instructions and caveats on cameras and similar ones in the  TZ series are mentioned. Now these are not m4/3 it seems so probably not supporting Bulb or M mode... Bulb for me is very important since i am doing all this for astrophotography.

quick quiestion:  is there a way to make -install only the lumix branch? i tried with the ONLY_THESE_CAMLIBS (or equiv) flag but still the install takes 2 minutes. i have given up on gdb and just print statements but the make process is quite time wasting. 

Lastly in tethered mode (USB) i see that GH5 is supported but GH5S is out and should also be working in exactly the same way (although resolution is different). I could test out a new release with GH5s/USB!

thnaks

totoantibes, 2019-06-09:
Marcus, 
the panasonic site was down yesterday. since then it is back up again. DC-TZ91 is indeed supported. 
as mentioned here: https://av.jpn.support.panasonic.com/support/global/cs/soft/image_app/dsc/android/index.html

full list of features supported by camera is avail here: https://av.jpn.support.panasonic.com/support/global/cs/soft/image_app/dsc/android/android02.html#gfg
the camlib should really spoof the image app behavior. with : 
- liveview streaming
-movie/still capture
-focus/aperture/zoom control
-Quality/EV/etc.

there is a complete C# implementation of all the features here: 
https://github.com/Rambalac/GMaster/blob/master/Core/Camera/Lumix.cs

msmeissn, 2019-06-09:
cd camlibs
make INSTALL_THESE_CAMLIBS="lumix.la" install

will only install the lumix driver, which should be shorter running.

cloudwheels, 2020-05-04:
I have a Lumix DMC G7 and have been tracking this for some time with the hope of using it for video streaming.
I'm pleased to confirm that I have actually achieved this with the libgphoto2 2.5.24 release, despite the release notes saying that capture is not yet possible.
Initially, having set up a wifi hotspot on the camera (menu > spanner >Wifi function> New connection >Remote Shooting & View),  the logs were showing an error which I had also experienced when trying to connect directly in a browser, which boils down to the camera expecting to connect to the authorised Image App:
```
<camrply>
<result>err_unsuitable_app</result>
</camrply>
```
However, after following [this solution](https://github.com/peci1/lumix-link-desktop/issues/8#issuecomment-608435677), I am able to authorise the camera and everything works (massive props to @peci1 and [his project](https://github.com/peci1/lumix-link-desktop):
> try manually opening http://192.168.54.1/cam.cgi?mode=accctrl&type=req_acc&value=0&value2=Lumix%20Link%20Desktop before running the app? This seems to be the magical command that opens the connection. It also forces the camera to ask you to authorize the device on first connection

I am assuming that `value2` could/should be changed to properly reflect that the requesting app is gphoto2.

Following that I was able to start a stream using:
```
gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0
```
I had previous installed the prerequisites for streaming with ffmpeg, as very clearly explained in [tis blog post](https://medium.com/nerdery/dslr-webcam-setup-for-linux-9b6d1b79ae22)

I built libphoto and gphoto2 following the instructions in the repos, which worked but are perhaps not too user friendly unless you're a developer. I found [these notes](https://sourceforge.net/p/gphoto/mailman/message/33367636/) to be very useful, cobmined with various google searches to fin the correct was to install the necessary xmllib2 and curllib headers. For this you could try (depending on your system):
```
sudo apt-get install -y libxml2-dev libcurl4-openssl-dev
```

Initial installation of the latest version of gphoto2 through the Ubuntu ppa gave me an outdated version, but according to the answer to [my question on ubuntu launchpad](https://answers.launchpad.net/ubuntu/+source/gphoto2/+question/690425) there is a ppa with version 2.5.24 (and seemingly other bleeding edge builds) available (subject to usual caution) maintained by Jasem Mutlaq: https://launchpad.net/~mutlaqja/+archive/ubuntu/libgphoto2

If I have some more time I will write this up a bit better, and ideally PR the necessary authorisation step, but in the meantime many thanks to @msmeissn and @totoantibes (and of course @peci1) for the sustained efforts of the years - I'm a very happy Lumix camera owner :)

dafontaine, 2020-05-18:
I can confirm the findings of @cloudwheels. Tested on my Lumix GF-7 using libgphoto2 2.5.24

willat8, 2020-05-20:
I can confirm the same for the GX9.
The Ubuntu/Debian libgphoto2-6 package has the lumix camlib disabled due to missing libcurl4-openssl-dev build dep. I have built with it enabled in this ppa https://launchpad.net/~willat8/+archive/ubuntu/libgphoto2 (only for groovy so far but could be backported).

msmeissn, 2020-05-20:
I have added the req_access command to the lumix driver too now.

willat8, 2020-05-22:
Hi @msmeissn I am proposing a small fix to e24b7e2 in #508.

msmeissn, 2020-05-22:
i added it to git, thanks!

willat8, 2020-05-24:
Some other observations (I am testing with GX9):

1. Picture quality is changed to RAW+JPEG every `camera_init`. I'm not sure about the reasons for this.
2. `--capture-movie` freezes after a variable amount of time, usually before or around 1 minute. I think this is due to the repeated `switchToRecMode (camera);` every `camera_capture_preview()` call

This commit [here](https://github.com/willat8/libgphoto2/commit/f6e9b1e026323100536ff1669fbb479f48ba7c5c) addresses these issues on the GX-9 but it'd be nice to hear from others before proposing this change here. (I've built the fixes in my PPA https://launchpad.net/~willat8/+archive/ubuntu/libgphoto2 for groovy).

totoantibes, 2020-05-24:
thanks,
1) the priority intention for this project for me was to capture RAW for astrophotography. Hence when I started the camlib i prioritized on RAW+Fine. there is no technical reason to force it to RAW+FINE but that makes sure we have all the available formats. If you can put another option in there to change the capture mode then please feel free.
2)  i have started a new branch that I believe handles better the preview and movie capture from the original work done with Marcus. It is not yet on any git (other than my local repo on my RPi) so for simplicity sake I am including it here. there is some diff/Merge to be done with latest master branch on libgphoto.
robert

cloudwheels, 2020-10-08:
@willat8 I've just come back to this afer some months and have the following error. Not exactly sure what is going on here, I'm just wondering if there's anything obvious before I start digging deep...
I haven't changed anything in the setup, but ran `pip3 install gphoto2` after intially gettng the error, in an attempt to update the library and then `sudo apt get gphoto2` and have also added your ppa (output of `gphoto2 --version at the bottom.)
Unfortunately I've totally forgootenb what I'm doing here and looking for a straightforward solution to get up and running again!

```
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
[mjpeg @ 0x557ff11ca780] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 320x184, 25 tbr, 1200k tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
[swscaler @ 0x557ff120fb40] deprecated pixel format used, make sure you did set range correctly
[video4linux2,v4l2 @ 0x557ff11cdac0] Unable to open V4L2 device '/dev/video0'
Could not write header for output file #0 (incorrect codec parameters ?): No such file or directory
Error initializing output stream 0:0 -- 
Conversion failed!
``` 

```
gphoto2 --version
gphoto2 2.5.23.1

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), no exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24         standard camlibs, gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial, gcc, ltdl, no EXIF, no USB, serial without locking
```

willat8, 2020-10-09:
Hey @cloudwheels, using
    gphoto2 --capture-movie --stdout | ffplay -i pipe:0 -vcodec mjpeg
I get a resolution of `640x480` with my GX9.
With the current state of master I don't get the frequent freezes, just a pause after a few minutes where the camera seems to go to sleep (but tapping the screen resumes it).

cloudwheels, 2020-10-09:
@willat8 thansk for taking time to read me convoluted comment!

I think the freezing was possibly related to my wifi adapter.

I'm still getting 320x184.

My intital hunch was that the resolution is related to the config value `gphoto2 --get-config /config/settings/liveviewsize`, but this reports as `qvga` (320x240). 320x184 isn't even a sensible aspect ratio!

Disappointingly, I'm sure I has a good resolution back in May. For the hours I spent, for my purposes (use as webcam) it's probably just easier to buy an elgato camlink 4k and be done with it :)

Still, any ideas on whether this is a camera limitation or something I might be able to alter are very welcome.

willat8, 2020-10-14:
The lumix camlib is now built by default in Debian libgphoto2 2.5.26-1 (and hopefully Ubuntu soon) https://salsa.debian.org/debian-phototools-team/libgphoto2/-/merge_requests/3

rmonat, 2020-10-26:
Out of curiosity, what is the maximum liveview resolution supported by the Panasonic G9/GH5? It appears that the Lumix Webcam Software on Windows/Mac is able to output up to 1280x720 streams.

totoantibes, 2020-10-26:
I believe it is VGA as per the Panasonic App modes. the lumix webcam would be over USB not WIFI.

iak92, 2021-03-07:
Hey,
I wanted to test my Lumix DMC G70 with gphoto. When connecting over ip: I get an error that only ptpip and usb are supported. When using ptpip: i can at least red abilities but can't take picture or use it as a webcam.  I always get am I/O Error

`gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
`
any recomandations? I also tried gtkam. There I can see the picture on the camera, but I can't take images. Also gtkam ist super slow.

svatamista, 2021-12-01:
Hi, I wanted to ask, especially @totoantibes Have you been succesful with running lumix GX9 with stellarmate OS? I am doing research, in order to be able start atrophotography, and GX9 is listed so far only as "experimental" on stellarmate OS help page.
Thanks for response.

Informatic, 2022-10-25:
Quick report from Lumix G80 user. (so that I don't forget to contribute here... :)

First `capture-image` makes the camera capture an image, but fails to "succeed" in gphoto2 shell. All other captures fail. This is due to call to `get_content_info`/`NumberPix` which returns `err_busy` unless camera is in `playmode` in here. I'll try patching up `lumix` camlib to switch to playmode early during `NumberPix` (or prevent that call altogether) sometime soon and test if this helps with this bug.

nmschulte, 2023-03-25:
Hi.  I'm new to this world, with a [Panasonic LUMIX DC-ZS70 (DC-TZ90)](https://www.shop.panasonic.com/cameras-and-camcorders/cameras/point-and-shoot-cameras/dc-zs70k), which has 802.11 b/g/n WiFi.  The manual indicates the camera can connect in a number of different ways, including to a smartphone app (https://play.google.com/store/apps/details?id=com.panasonic.avc.cng.imageapp), to a PC (running e.g. PHOTOfunSTUDIO, or not), and various "web services" including "LUMIX CLUB."

I'm looking for a way to transfer content without having to connect to the device via USB, and it seems like this may be it, but I've been unsuccessful w/ stock builds of gphoto2.  Can someone school me about this device, and what I should be looking to for this ability?  Is the `lumix` branch here it, is more work needed, or no/otherwise?

Thanks!

passteinforth, 2024-10-07:
Hey,
I just found this and maybe someone ( @cloudwheels or @willat8 ?) can answer my question.
I try to connect my GX9 to gphoto via wifi and use it for a photobooth. USB connection is recognized, but I can't take pictures with it. However, I can take pictures via Lumix Link Desktop (no stream though), but I don't know how to get gphoto2 using this connection. 
Is it possible to control the GX9 with gphoto2 via Wifi?
I am currently on  gphoto2 v2.5.28, libgphoto2 v2.5.30 and libgphoto2_port 0.12.1

Thanks!