gphoto/libgphoto2 #411: Running gphoto2/libgphoto2 on Windows IoT Core (or Desktop UWP app) Error : 'Out of memory' (-3)

mms-, 2019-04-27:
Running gphoto2 mingw 64 bit binaries on Windows IoT Core and unable to detect or load the usb library. 

Reproduce using : gphoto2.exe --debug --debug-logfile=my-logfile.txt --auto-detect --summary

The same binaries on desktop Windows 10 work just fine. However this can be reproduced on Windows 10 desktop, the exact same error occurs when using libgphoto2 library directly from a UWP app. Faults at the same point with same output, trying to load the usb1 iolib.

Seems like it is unable to access or enumerate usb devices and the error itself is misleading?

This occurs with or without winusb driver installed for the Nikon camera and weather it is connected or not. Seems like a general failure. Not related to a camera. Any ideas?

Libusb is the cause or the error:

gp_port_library_list libusb1.c:216: Out of memory: 'descs = calloc (nrofdevs, sizeof(descs[0]))' failed.

Complete output:

0.000000 main (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.015621 main (2): gphoto2 2.5.20
0.015621 main (2): gphoto2 has been compiled with the following options:
0.031240 main (2): + x86_64-w64-mingw32-gcc (C compiler used)
0.031240 main (2): + popt (mandatory, for handling command-line parameters)
0.031240 main (2): + exif (for displaying EXIF information)
0.046862 main (2): + no cdk (for accessing configuration options)
0.046862 main (2): + no aa (for displaying live previews)
0.062481 main (2): + jpeg (for displaying live previews in JPEG format)
0.062481 main (2): + readline (for easy navigation in the shell)
0.062481 main (2): libgphoto2 2.5.21
0.062481 main (2): libgphoto2 has been compiled with the following options:
0.062481 main (2): + all camlibs
0.062481 main (2): + x86_64-w64-mingw32-gcc (C compiler used)
0.062481 main (2): + ltdl (for portable loading of camlibs)
0.062481 main (2): + EXIF (for special handling of EXIF files)
0.062481 main (2): libgphoto2_port 0.12.0
0.078060 main (2): libgphoto2_port has been compiled with the following options:
0.078060 main (2): + iolibs: disk ptpip usb1
0.078060 main (2): + x86_64-w64-mingw32-gcc (C compiler used)
0.078060 main (2): + ltdl (for portable loading of iolibs)
0.078060 main (2): + USB (libusb1, for USB cameras)
0.078060 main (2): + no serial (for serial cameras)
0.078060 main (2): + no resmgr (serial port access and locking)
0.078060 main (2): + no ttylock (serial port locking)
0.078060 main (2): + no lockdev (serial port locking)
0.078060 main (2): CAMLIBS env var not set, using compile-time default instead
0.093678 main (2): IOLIBS env var not set, using compile-time default instead
0.093678 main (2): invoked with following arguments:
0.093678 main (2): --debug
0.093678 main (2): --debug-logfile=my-logfile.txt
0.093678 main (2): --auto-detect
0.093678 main (2): --summary
0.109305 load_settings (2): Creating gphoto config directory ('C:\Data\Users\administrator.gphoto')
0.109305 verify_settings (2): Can't open settings file 'C:\Data\Users\administrator.gphoto\settings' for reading.
0.109305 load_settings (2): Loading settings from file 'C:\Data\Users\administrator.gphoto\settings'.
0.109305 load_settings (2): Can't open settings file 'C:\Data\Users\administrator.gphoto\settings' for reading.
0.109305 load_settings (2): Creating gphoto config directory ('C:\Data\Users\administrator.gphoto')
0.124925 verify_settings (2): Can't open settings file 'C:\Data\Users\administrator.gphoto\settings' for reading.
0.124925 load_settings (2): Loading settings from file 'C:\Data\Users\administrator.gphoto\settings'.
0.124925 load_settings (2): Can't open settings file 'C:\Data\Users\administrator.gphoto\settings' for reading.
0.124925 main (2): The user has not specified both a model and a port. Try to figure them out.
0.124925 gp_port_info_list_load (2): Using ltdl to load io-drivers from '.\IOLIBS'...
0.124925 foreach_func (2): Called for filename './IOLIBS/disk'.
0.140543 foreach_func (2): Loaded '' ('^disk:') from './IOLIBS/disk'.
0.140543 foreach_func (2): Called for filename './IOLIBS/ptpip'.
0.140543 foreach_func (2): Loaded 'PTP/IP Connection' ('ptpip:') from './IOLIBS/ptpip'.
0.140543 foreach_func (2): Loaded '' ('^ptpip:') from './IOLIBS/ptpip'.
0.140543 foreach_func (2): Called for filename './IOLIBS/usb1'.
0.186573 gp_port_library_list libusb1.c:216: Out of memory: 'descs = calloc (nrofdevs, sizeof(descs[0]))' failed.
0.202190 foreach_func gphoto2-port-info-list.c:237: Error during assembling of port list: 'Out of memory' (-3).
0.202190 foreach_func (2): Loaded '' ('^usb:') from './IOLIBS/usb1'.
0.202190 gp_port_info_list_count (2): Counting entries (4 available)...
0.202190 gp_port_info_list_count (2): 1 regular entries available.
0.202190 gp_abilities_list_load_dir (2): Using ltdl to load camera libraries from '.\CAMLIBS'...
0.202190 foreach_func (2): Found './CAMLIBS/adc65'.
0.202190 foreach_func (2): Found './CAMLIBS/agfa_cl20'.
0.217810 foreach_func (2): Found './CAMLIBS/aox'.
0.217810 foreach_func (2): Found './CAMLIBS/ax203'.
0.217810 foreach_func (2): Found './CAMLIBS/barbie'.
0.217810 foreach_func (2): Found './CAMLIBS/canon'.

msmeissn, 2019-04-28:
we discussed this on the mailinglist ... I cant help much with windows :/

peterbud, 2019-04-28:
If you read the [Windows IoT Core Overview from Microsoft ](https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-core) it's clearly stating:

> Differences in driver-supported areas

>    Windows 10 Desktop has more supported drivers than Windows 10 IoT Core. To make the same device(s) work on Windows 10 IoT Core as on Desktop, you may need to build a driver from soruce for a Windows 10 IoT Core device or find another workaround, especially for ARM architecture.
>    There is no out-of-the-box driver for libusb for Windows 10 IoT Core (ARM) - you will need to build from source to target the ARM architecture.

I'm not really sure why you are even trying to run libgphoto2 on a Windows IoT core machine, would you mind explaining the reason?

mms-, 2019-04-28:
This is amd64 so the driver is included? What else is needed to make libusb have access and be able to enumerate USB devices on IoT Core? Also this isn't just an IoT Core issue, the same thing happens with a UWP process on full Windows 10, yet it works in a .Net Core 2.0 process. Can be reproduced with mingw libgphoto2 package and a basic c# wrapper.

The error message as @msmeissn noted is due to nrofdevs likely being 0, and erroneously considering the NULL return as a bug.

Would like to use libgphoto2 with DSLR cameras on a IoT Core device.

driver1998, 2019-07-09:
You'll still need to install the libusb driver even if you are using x64 IoT Core.
Maybe use devcon to install it.
Also, IoT Core require the use of Universal drivers (basically a type of driver with relatively fewer APIs and works on IoT Core/Mobile/Desktop/whatever).

mms-, 2019-07-09:
I have not yet gotten the driver compiled and installed for IoT Core. However this doesn't answer the question why it fails the same way in a UWP processes on full Windows 10 but works in a .Net Core 2.0 process. The driver is installed.