gphoto/libgphoto2 #413: Bulb mode for Canon EOS M50

antocuni, 2019-05-09:
Bulb mode does not seem to be supported on Canon EOS M50, even by the latest git version (I have tried with 0c45b3615).

If you try to run `gphoto2 --capture-image -B 1` you get this ([logfile](https://github.com/gphoto/libgphoto2/files/3162341/my-logfile.txt))
```
$ gphoto2 --capture-image -B 1
Bulb mode enabled (exposure time: 1s).                                         

*** Error ***              
bulb not found in configuration tree.
ERROR: Could not set bulb capture, result -1.
*** Error (-1: 'Unspecified error') ***       
```

I tried to dig a bit in the code. If I understood things correctly, bulb mode for Canon cameras is implemented by `PTP_OC_CANON_EOS_BulbStart`, but my M50 does not seem to support it. I even tried to force the inclusion of `actions/bulb` in the prop list, but as expected the camera responds with this when I try to use it:
```
*** Error ***              
PTP Operation Not Supported
```

I am willing to contribute a patch to have it working, but I need some guidance. I think there are at least two ways to do it:
1. find the proper PTP command which works on the M50; however, I have no idea how to search for it
2. implement bulb mode by sending `Canon EOS Remote Release` commands: I tried to do it manually and it seems to work, but you need to manually set the shutter speed to `Bulb` before doing it.

msmeissn, 2019-05-09:
check out the "eosremoterelease" option ... see http://gphoto.org/doc/remote/

antocuni, 2019-05-09:
> check out the "eosremoterelease" option ... see http://gphoto.org/doc/remote/

Thank you, this is useful. Am I right to assume that it's not integrated with `--capture-image -I` though? Just checking, it's not really a probably as I can easily script it.

msmeissn, 2019-05-09:
its hard to integrate there, but I see if I can add it in

RReverser, 2023-07-02:
FWIW same issue on EOS 250D. `gphoto2 --bulb=...` doesn't work with "(0): bulb not found in configuration tree." in the logs. I'll give `eosremoterelease` a try, but would be nice to have all those options unified somehow. (this leads back into #853)