gphoto/libgphoto2 #414: gphoto2.GPhoto2 Error: [-107] Directory not found

uvedhe, 2019-05-10:
**Describe the bug**

When the camera takes long exposure shots (~30 seconds), gphoto2 returns `gphoto2.GPhoto2 Error: [-107] Directory not found` after capture has finished (shutter closes). Running the exact same script for exposures that are shorter does not result in this error.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

uvedhe, 2019-05-10:
Actually, opened an issue at gphoto2 also, because probably more to do with that than with libgphoto(?)

msmeissn, 2019-05-19:
There is max limit of 35 seconds sony capture will wait for a picture to be added, then itz will error out..

for longer wait times, use trigger capture and wait_For_event