gphoto/libgphoto2 #416: PTP I/O Error

manojtiwary22, 2019-05-24:
Hi,

I'm trying to Parse the list of files present in my camera to my debian jessie device using gphoto2 -L command , repetitively in a while loop for 500 iteration. But the loop stops at around  
60-100th iteration with gp_camera_folder_list_files() returning -1 and following error:

*** Error ***              
PTP I/O error
29.294883 gphoto2-port(2): Closing port...
29.297928 context(0): An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-1: 'Unspecified error') ***       

output of : gphoto2 --auto-detect 
USB PTP Class Camera           usb:002,003

output of: gphoto2 --version
gphoto2 2.5.4
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.4          gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.4          all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.10.0         gcc, ltdl, no USB, serial without locking

Steps to reproduce the behavior:
I'm just repetitively calling gphoto2 -L for 500 iterations.
 
I've attached my debug file.
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/3216082/my-logfile.txt)

I don't know whether it is a issue with camera or my device because the same thing works perfectly fine in my ubuntu PC. But when I debug the transaction with USB analyzer I will get NAK for host's command block.
here is a screenshot of the usb analyzer
[image]
 

Thanks for the help.

msmeissn, 2019-06-29:
This seems to happen at hardware level, not in the realm of libgphoto2.

not sure if it is the ARM device you are running on or the target device.

there is not much possible for libgphoto2 I think.

nicolassilvar, 2019-08-11:
I'm having the samme isue. 

An error occurred in the io-library ('Unspecified error'): **The supplied vendor or product id (0x0,0x0) is not valid.**

ArchCarbon% gphoto2 -l
                                                                               
*** Error ***              
PTP General Error

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-1: 'Unspecified error') ***   

ArchCarbon% gphoto2 -a                
Abilities for camera             : USB PTP Class Camera                        
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Image
                                 : Preview
Configuration support            : yes
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes

ArchCarbon% gphoto2 -v
gphoto2 2.5.23

From the log file attached, it seems there is an Error Writing to the port, but does not seem to come from USB permission

0.215348 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x1002 sending req failed: Error writing to the port (-35)
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/3490409/my-logfile.txt)

nicolassilvar, 2019-08-11:
Using Camera SONY DSC-RX100 III