gphoto/libgphoto2 #417: Cross-compile the example for ARM development board

cxd-nanning, 2019-05-29:
Hi all,
I cross-compile the example and download it to the ARM development board to run, but the compilation error:
root@ubuntu:/home/cxd/app/liu/gphoto/libgphoto2-2.5.4/examples# arm-hisiv300-linux-gcc sample-capture.c -o sample-capture2 -I/home/cxd/app/arm-install/include/ -L/home/cxd/app/arm-install/lib/
/tmp/ccVQeMBw.o: In function capture_to_memory': sample-capture.c:(.text+0xb4): undefined reference to gp_camera_capture'
sample-capture.c:(.text+0xfc): undefined reference to gp_file_new' sample-capture.c:(.text+0x14c): undefined reference to gp_camera_file_get'
sample-capture.c:(.text+0x170): undefined reference to gp_file_get_data_and_size' sample-capture.c:(.text+0x1a8): undefined reference to gp_camera_file_delete'
/tmp/ccVQeMBw.o: In function capture_to_file': sample-capture.c:(.text+0x248): undefined reference to gp_camera_capture'
sample-capture.c:(.text+0x2a8): undefined reference to gp_file_new_from_fd' sample-capture.c:(.text+0x2f8): undefined reference to gp_camera_file_get'
sample-capture.c:(.text+0x340): undefined reference to gp_camera_file_delete' sample-capture.c:(.text+0x35c): undefined reference to gp_file_free'
/tmp/ccVQeMBw.o: In function main': sample-capture.c:(.text+0x390): undefined reference to sample_create_context'
sample-capture.c:(.text+0x3a4): undefined reference to gp_log_add_func' sample-capture.c:(.text+0x3b0): undefined reference to gp_camera_new'
sample-capture.c:(.text+0x3c8): undefined reference to gp_camera_init' sample-capture.c:(.text+0x4a0): undefined reference to gp_camera_exit'
collect2: error: ld returned 1 exit status

what should I do?
thanks!

msmeissn, 2019-05-29:
add -lgphoto2 -lgphoto2_port   to the commandline above.

cxd-nanning, 2019-05-30:
Thanks Mr msmeissn, I will try now.

cxd-nanning, 2019-05-31:
Thanks Mr msmeissn, When I install libgphoto2-2.5.22 and add -lgphoto2 -lgphoto2_port to  the commandline above. Cross-compile and download it to the ARM development board to run is right. thank you very much!

closed 2019-05-31 by cxd-nanning