gphoto/libgphoto2 #418: Nikon Z6 movie support

avisecag, 2019-06-07:
Hi

I tried to get a movie captured with a Nikon Z6. In gPhoto2 command line application I do not get any errors but neither a get a result. Tried it with this command:
`gphoto2 --set-config movie=1 --wait-event=10s --set-config movie=0 --wait-event-and-download=2s`

And putting the camera to viewfinder mode does not help.

It is possible to captured tethered movie with Nikon's Camera Control Pro. But I am not able to filter a USB traffic on ptp commands. How do you achieve that @msmeissn ?

Thanks a lot.

[nikonZ6MovieDebug.log](https://github.com/gphoto/libgphoto2/files/3264814/nikonZ6MovieDebug.log)

avisecag, 2019-06-12:
Ok, we're in a weird place right now.

As mentioned above, the reason why we can't capture video is that we can't set applicationmode to 1. We get 0x200f PTP Access denied back when we try to do that.
However, another developer has entered the fray to try to figure it out, and it worked without issue on his machine.
After a bit of frantic testing on various hardware, that HP laptop running Ubuntu 18 still has remained the only machine on which applicationmode can be set (and as a result of that, video can be captured).
We tried other Ubuntu 18 machines, as well as a raspberry pi 3B with raspbian stretch... On all of them, the Z6 (nor any other nikon with electronic shutter as far as we can tell) will allow the operation. Nikon models with normal shutters seem to do fine.

The operation *can* be performed on those machines on windows using both camera control pro or digicam control, but not using gphoto under linux.

Attached are two files that show the debug trace of the operation failing on the pi,  and it succeeding on an HP laptop. We currently have no idea where the issue might be originating. We're not even sure what gphoto's part in this is. I'm mostly posting this in the hopes that somebody can provide any pointers as to how to fix the problem.
[appmode-set-success.txt](https://github.com/gphoto/libgphoto2/files/3281890/appmode-set-success.txt)
[appmode-set-fail.txt](https://github.com/gphoto/libgphoto2/files/3281891/appmode-set-fail.txt)

avisecag, 2019-06-13:
We got it working, but we're not really sure what the problem was.

At some point we noticed that capturing movies worked if gphoto2 was installed as apt package, but when installed via the gphoto2-installer it produced the above error.
Since the package for the pi is hopelessly outdated, this wasn't quite the end of our worries. But eventually we figured out that if we installed just the libgphoto2-port12 package additionally to what gphoto2-updater installed, it worked.

gphoto2 --version still shows the same version (0.12.0), but with a slightly shorter description, so it doesn't seem to be *quite* the same version? Not sure. In any case, this makes this issue more related with the gphoto2-updater than with gphoto2 itself.

closed 2019-06-13 by avisecag

msmeissn, 2019-06-13:
gphoto updater just installs the latest libgphoto2 and gphoto2 version, so it seems the newer version broke it... limited to port12 is weird though, we have done some changes to the libusb1 driver but not much

dvergeylen, 2020-02-26:
We are experiencing the exact same problem. We can reproduce it reliably. `libgphoto2_port.so.12` is needed to work correctly.

```
$ ldd /usr/bin/gphoto2
	linux-vdso.so.1 (0xbed00000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f10000)
	libgphoto2.so.6 => /usr/lib/arm-linux-gnueabihf/libgphoto2.so.6 (0xb6e68000)
	libgphoto2_port.so.12 => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so.12 (0xb6e4e000)
	libcdk.so.5 => /usr/lib/arm-linux-gnueabihf/libcdk.so.5 (0xb6e0c000)
	libncurses.so.6 => /lib/arm-linux-gnueabihf/libncurses.so.6 (0xb6dde000)
	libaa.so.1 => /usr/lib/arm-linux-gnueabihf/libaa.so.1 (0xb6db3000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabihf/libjpeg.so.62 (0xb6d6e000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6d44000)
	libexif.so.12 => /usr/lib/arm-linux-gnueabihf/libexif.so.12 (0xb6d01000)
	libreadline.so.7 => /lib/arm-linux-gnueabihf/libreadline.so.7 (0xb6cb5000)
	libpopt.so.0 => /usr/lib/arm-linux-gnueabihf/libpopt.so.0 (0xb6c9a000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6c18000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6aca000)
	/lib/ld-linux-armhf.so.3 (0xb6f56000)
	libltdl.so.7 => /usr/lib/arm-linux-gnueabihf/libltdl.so.7 (0xb6ab2000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0xb6a81000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6a6e000)
	libslang.so.2 => /lib/arm-linux-gnueabihf/libslang.so.2 (0xb685f000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabihf/libX11.so.6 (0xb673d000)
	libgpm.so.2 => /usr/lib/arm-linux-gnueabihf/libgpm.so.2 (0xb6727000)
	libxcb.so.1 => /usr/lib/arm-linux-gnueabihf/libxcb.so.1 (0xb66f8000)
	libXau.so.6 => /usr/lib/arm-linux-gnueabihf/libXau.so.6 (0xb66e5000)
	libXdmcp.so.6 => /usr/lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb66d0000)
	libbsd.so.0 => /usr/lib/arm-linux-gnueabihf/libbsd.so.0 (0xb66a8000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6691000)
```