gphoto/libgphoto2 #419: Add Sony DSC RX0 II

sops21, 2019-06-07:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Sony DSC RX0 II

**USB IDs**
Bus 001 Device 006: ID 054c:0ca6 Sony Corp. 

**camera summary output**

[summary.txt](https://github.com/gphoto/libgphoto2/files/3266562/summary.txt)

**camera configuration output**

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/3266565/list-all-config.txt)

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
Does not work consistently. Sometimes it works and downloads. Other times it just sits and does not capture. Log file is to big to attach
[Log.log](https://drive.google.com/file/d/1Qsw5NW0f6i8nmhclCPrCIJzE2RvX3abj/view?usp=sharing)
  gphoto2 --capture-preview
Capture Preview works and downloads a .jpg

msmeissn, 2019-06-09:
i added the id to our git ... hmm

sops21, 2019-06-15:
Thanks for adding that. What is the best way for me to test. Do I need to compile from the GIT repo?

sops21, 2019-06-21:
It seems that I can capture video when in "Computer Control" and download video when in MTP mode. Do you think there is a way to Record and then download the video? (without physically switching the modes on the camera)?

msmeissn, 2019-06-23:
in the older sony a58 i have this is not possible (capture and download movie in the same mode). not sure why.

It seems you got it running! :)

sops21, 2019-06-23:
I can get the 2 commands working if I use the 2 different modes, for my project I need to be able to record and download. Do you know any why to switch modes programmaticly between computer control and MTP?

msmeissn, 2019-06-30:
I currently do not know. It might be possible though, I just dont know.

romainreignier, 2019-07-03:
Thank you @sops21 for the RX0 II support.

I am currently testing it but do not manage to set the Exposure Compensation?
Did you try and manage to do it?
On the RX0 I (firmware 3.00), it was working but not on RX0 II (firmware 1.00).

What is strange is that there is no choices proposed:
```
$ gphoto2 --get-config=exposurecompensation
Label: Exposure Compensation                                                   
Readonly: 0
Type: RADIO
Current: -2
END
```
If I try to set it:
```
$ gphoto2 --set-config=exposurecompensation=-1.0
```
The compensation do change but to 0. Whatever setting I try, it goes to 0.

And it is the same using directly the PTP code `5010`:
```
$ gphoto2 --get-config=5010
Label: Exposure Bias Compensation                                              
Readonly: 0
Type: MENU
Current: -2000
END
```
But in that case, the compensation do not change on the camera after:
```
$ gphoto2 --set-config=5010=-1
```

@msmeissn Do you have any idea?

Maybe we should wait for a new release of the firmware?

msmeissn, 2019-10-12:
sorry for replying late... for the exposure bias problem a debug logfile of the change attempt would be good.

romainreignier, 2019-10-13:
@msmeissn Thank you for your answer. But I do not have the RX0 II at hand anymore. 
For now I am sticking on the first version.
Sorry about not having the logs.

msmeissn, 2019-10-20:
Okay ... no problem. If you can get data, please open a new ticket :)

closed 2019-10-20 by msmeissn