gphoto/libgphoto2 #420: 2.5.23, Nikon cameras fail to take in low light.

mr-russ, 2019-06-24:
**Describe the bug**

2.5.23, Nikon cameras fail to take in low light.

**Name the camera**

Nikon D7200
Nikon D5600
Nikon D850

**libgphoto2 and gphoto2 version**

Was libgphoto only in C++ program. It was a master from 4th Feb.  I have confirmed the following commit is the timeout that casuses the issue.

https://github.com/gphoto/libgphoto2/commit/6929c05bec27d07808b67388526cb53f97183021#diff-b77be88023d465cda8bd7c00af114c50R5214

**To Reproduce**

We have a number of reports of this issue, but below is how we reproduced it on the bench to verify the issue and fix.

Get a set of post-it notes and place them over the front of the lense, ensuring that focus will take > 5 seconds.  This is done with a small angle available for light at the bottom, not flat against the lense.

* Set the camera to Aperture priority 
* Take a photo

https://github.com/gphoto/libgphoto2/commit/6929c05bec27d07808b67388526cb53f97183021#diff-b77be88023d465cda8bd7c00af114c50R5214
Show that an infinite timeout was replaced with 5 seconds of timeout.  That 5 seconds is not always long enough.

```
camera_nikon_capture [library.c:3480] 'nikon_wait_busy (params, 100, 5000)' failed: 'PTP Device Busy' (0x2019)
camera_nikon_capture [library.c:3480] 'nikon_wait_busy (params, 100, 5000)' failed: 'PTP Device Busy' (0x2019)
gp_context_error PTP Device Busy
gp_context_error PTP Device Busy
gp_camera_capture [gphoto2-camera.c:1340] 'camera->functions->capture (camera, type, path, context)' failed: -110
gp_camera_capture [gphoto2-camera.c:1340] 'camera->functions->capture (camera, type, path, context)' failed: -110
```

msmeissn, 2019-06-24:
any idea how long it at most will take? 10 seconds? 20 ?

mr-russ, 2019-06-27:
I've seen very close to 10 in my testing.  I would be going for 20 if you are increasing the timeout.

Ideally passing in a total time limit and decreasing that.  But my reading of the code says that will be really complicated given the range of cameras and methods they use to take photos.

msmeissn, 2019-06-29:
i bumped the limits to 20, also activating liveview to 2 seconds.

(Main goal was to get rid of infinite loops)

closed 2019-06-29 by msmeissn