gphoto/libgphoto2 #421: Canon EOS 1200D Screen goes black

saraara, 2019-06-25:
I have strange issue while capturing image after every 5 seconds automatically. After couple of dozens images being captures correctly, somehow camera turns the screen to black, then camera hangs & takes sometime, and then error out without capturing image. The issue seems to be with SD card. Camera seems unable to capture image to SD card. When turning off and on the camera using the top dial switch, the camera start functioning fine again. but then this issue repeats after several dozen images being captured.

Any help fixing this issue?

The command I issue from command prompt is --capture-image-and-download

Mahmood-Hussain, 2019-06-25:
Camera takes sometime to transfer images to SD card. Capturing images at an
interval of 5 seconds maybe the main cause for this problem (maybe buffer
is full after sometime and then it’s unable to do further operations). Try
to increase 5 seconds to 30 seconds and note the output it should work.

saraara, 2019-06-25:
I actually increased the interval from 5 seconds to 40 seconds, and this problem remains.

In another words I first started capturing image at every 5 seconds, and encounter above issue.
I then gradually increased time interval unto 30 seconds, and above issue remains.

It works just fine for several dozen image capture, and then somehow camera lost connection with SD card. turning camera power off and on makes everything to work just fine until next time this issue reappears.

Mahmood-Hussain, 2019-06-25:
Which camera you are using?
Are you using HDD(with like raspberry Pi) or SD camera inside camera?

saraara, 2019-06-25:
Camera straight connected with USB cable to Mac. Camera has SD card inside. I am capturing images through simple script. That's about it.

Mahmood-Hussain, 2019-06-25:
Which camera?

saraara, 2019-06-25:
Canon EOS 1200D

Mahmood-Hussain, 2019-06-25:
This issue seems to have links with camera and SD card.
(If your computer switches from standby to sleep that may also cause this
problem. Try to increase the standby time of your Mac)

saraara, 2019-06-25:
Mac set-up for never turn the screen off or never goes into sleep mode. Thought something beyond this normal troubleshooting might help to resolve this issue.

Mahmood-Hussain, 2019-06-25:
I used Canon EOS 550D with python script to automatically capture images
that worked fine for me.

msmeissn, 2019-06-30:
What is
 gphoto2 --get-config capturetarget 
showing?
If this only happens after quite a number of capture to card, it might also be a faulty card, can you replace it with a fresh one?