gphoto/libgphoto2 #422: set copyright error  using canon eos r camera

jadenjiang, 2019-06-27:
**Describe the bug**
    I  want to set copyright using libgphoto2, but it's failed and camera will restart.  i find that  if the setting type is text like ownername, artist, setting action is awaly failed, but  if the setting type is date and then setting action will success.
    if you push camera's menu button let camera show setting desktop, and then setting copyright action is success.

**Name the camera**

Canon EOS R            
ID 04a9:32da Canon, Inc.

**libgphoto2 and gphoto2 version**

gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**

set copyright alway failed when don't push camera's menu button.

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

[debug.log](https://github.com/gphoto/libgphoto2/files/3332759/debug.log)

msmeissn, 2019-07-02:
I tried with EOS M10 and EOS 750D, but it works there ... :/

jadenjiang, 2019-07-02:
thanks for your reply.
    you are right, not all canon cameras have this problem, like EOS 750D, EOS 5D Mark IV  and so on, they performance very well when using libgphoto2. 
    you can use EOS R for test if that in you hand and you will find that setting copyright action will failed when camera is not menu mode.   could you give me some advise for solve this problem?

konstantinpavlikhin, 2019-07-03:
Hi! I develop a commercial app that among other features allows users to customize owner, author and copyright settings of their cameras and I also ran into this issue with EOS R and similar models. My PTP code is fully custom and doesn't depend on libgphoto, but I believe the way we set these fields is the same. My testing shows that EOS RP camera hangs reliably when being passed a `eosSetDevicePropertyValueEx` (`0x9110`) command with `eosOwner` property code (`0xD115`) and a string payload.

I've installed a Canon EOS Utility 3 to try to customize these fields within the official app and noticed that when I apply changes from my Mac the camera screen dims for about a second. I set up a PTP channel monitoring and found out that EOS Utility 3 always sets a UI Lock when owner, author and copyright settings are being altered.

Here is the sample of PTP log where I change owner name to "Konstantin Pavlikhin":
```
Command(code: 0x911B, parameter1: 1)
Response(code: ok)

Command(code: eosSetDevicePropertyValueEx)
  out: PTPData(code: 0x9110, nestedData: <1d000000 15d10000 4b6f6e73 74616e74 696e2050 61766c69 6b68696e 00>)
Response(code: ok)
  in: PTPData(code: 0x9110, nestedData: <1d000000 15d10000 4b6f6e73 74616e74 696e2050 61766c69 6b68696e 00>)

Command(code: 0x911C)
Response(code: ok)
```

Contrary to this, EOS Utility 2 doesn't set a UI lock when applying changes to my EOS 5D Mark II.

Hope this makes sense! It would be great to determine which cameras require this temporary UI lock or come with a generic approach like requesting it unconditionally for all EOS cameras.

msmeissn, 2019-07-03:
thanks for this info!

this would be largely equivalent to:

gphoto2 --set-config uilock=1 --set-config copyright="test" --set-config uilock=0

jadenjiang, 2019-07-04:
thanks  @msmeissn  @konstantinpavlikhin  for your help.
I will try this. I think the issue can be closed now.

msmeissn, 2019-07-04:
reopen ... if this workaround fixes rthe problem I need to adjust the code to do locking / unlocking