gphoto/libgphoto2 #423: Get picture path on Canon EOS RP

Patrick-Lascombe, 2019-06-27:
I'm using the canon EOS RP to take several pictures during a routine.

All the pictures are saved on the camera, to avoid the download time and we are taking the picture using the command set-config /main/actions/eosremoterelease=5 so we can take several pictures in a row as fast as possible.

I was wondering if there was a fast and simple way to know the path of the last picture taken so I can list them and download all of them later.

Thanks for your help and have a great day

msmeissn, 2019-06-27:
if you do --wait-event after this triggering (in same gphoto2 commandline) then you will get FILEADDED events for all the new pictures.

you can also list pictures and retrieve the alphabetically "n" last images, as the camera only adds to the end of the directory.

Patrick-Lascombe, 2019-07-02:
Thanks for your help it was very helpful ! 
I'm looking to recover the path of every picture taken with the cpp code below. I'm looking at all the event incoming until the right one is arrived. The problem is that it is time consuming (~1.3 seconds for every pictures).

Is there a way to get quicker the event GP_EVENT_FILE_ADDED ?
Thanks again for your help

```
CameraEventType evttype = GP_EVENT_UNKNOWN;
CameraFilePath * path;
 void *data = NULL;

while (evttype != GP_EVENT_FILE_ADDED) {
    int res = gp_camera_wait_for_event(camera_, 100, &evttype, &data, context_);
}

path = static_cast<CameraFilePath*>(data);
std::string complete_path = path->folder + std::string("/") + path->name;

return complete_path;
```

msmeissn, 2019-07-06:
the loop you use seems to be correct in regards to good polling.

likely the 1.3 seconds are taken to write to the CARD and the camera waits until its written to report it... :/

Patrick-Lascombe, 2019-07-08:
I found a workaround.
I'm using a vector of enum (tasks) that stores the actions wanted. After each actions or when no actions is required we listen to the events.
This program is running in a 50Hz loop, so it's still able to listen to the incoming interruption that add tasks to the vector.
With this setup I can take picture at 5Hz while still recovering the path to the image later on.

```
      if(tasks_.size() > 0) {
          switch(tasks_[0]) {
              case set_focus:
                  std::cout << "Task set focus" << std::endl;
                  setFocus();
                  break;

              case trigger_capture:
                  std::cout << "Task trigger capture" << std::endl;
                  triggerCapture();
                  break;

              case unlock_camera:
                  std::cout << "Task unlock camera" << std::endl;
                  unlockCamera();
                  break;

              case download_pictures:
                  std::cout << "Task download pictures" << std::endl;
                  downloadPictures();
                  break;

        }
        tasks_.erase(tasks_.begin());
      }

    recoverPath();
```

So this issue can be closed if you don't have anything to add.

Thanks for your help !

closed 2019-07-12 by Patrick-Lascombe