gphoto/libgphoto2 #425: memory leak  run gphoto2  --auto-detect

jadenjiang, 2019-07-02:
**Describe the bug**

memory leak  run gphoto2  --auto-detect

**libgphoto2 and gphoto2 version**
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

I use valgrind tool to check memory leak(valgrind --leak-check=full  /opt/libgphoto2/bin/gphoto2  --auto-detect)
the output is below:

==14041== HEAP SUMMARY:
==14041==     in use at exit: 107,340 bytes in 25 blocks
==14041==   total heap usage: 8,338 allocs, 8,313 frees, 83,954,011 bytes allocated
==14041== 
==14041== 4,096 bytes in 1 blocks are definitely lost in loss record 10 of 14
==14041==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14041==    by 0x76D870B: ???
==14041==    by 0x76DBB20: ???
==14041==    by 0x74C9C0C: ???
==14041==    by 0x74C8452: ???
==14041==    by 0x74BFC9C: ???
==14041==    by 0x72B2BEA: ???
==14041==    by 0x50CCBAD: foreach_func (gphoto2-port-info-list.c:233)
==14041==    by 0x5D39899: ??? (in /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1)
==14041==    by 0x5D3923B: ??? (in /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1)
==14041==    by 0x5D39E18: lt_dlforeachfile (in /usr/lib/x86_64-linux-gnu/libltdl.so.7.3.1)
==14041==    by 0x50CCD91: gp_port_info_list_load (gphoto2-port-info-list.c:285)
==14041== 
==14041== 4,096 bytes in 1 blocks are definitely lost in loss record 11 of 14
==14041==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14041==    by 0x76D870B: ???
==14041==    by 0x76DBDDB: ???
==14041==    by 0x76E011A: ???
==14041==    by 0x76E0580: ???
==14041==    by 0x76E25EB: ???
==14041==    by 0x76E29B4: ???
==14041==    by 0x76E3FDE: ???
==14041==    by 0x74C9C2A: ???
==14041==    by 0x74C8452: ???
==14041==    by 0x74BFC9C: ???
==14041==    by 0x72B2BEA: ???
==14041== 
==14041== 8,192 bytes in 2 blocks are definitely lost in loss record 12 of 14
==14041==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14041==    by 0x76D870B: ???
==14041==    by 0x76DBB20: ???
==14041==    by 0x74C9C0C: ???
==14041==    by 0x74C8452: ???
==14041==    by 0x74BFC9C: ???
==14041==    by 0x72B37B6: ???
==14041==    by 0x50CE943: gp_port_init (gphoto2-port.c:118)
==14041==    by 0x50CEE2A: gp_port_set_info (gphoto2-port.c:199)
==14041==    by 0x4E430A8: gp_abilities_list_detect (gphoto2-abilities-list.c:424)
==14041==    by 0x10F526: auto_detect_action (actions.c:658)
==14041==    by 0x11A52E: cb_arg_run (main.c:1580)
==14041== 
==14041== 8,192 bytes in 2 blocks are definitely lost in loss record 13 of 14
==14041==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==14041==    by 0x76D870B: ???
==14041==    by 0x76DBDDB: ???
==14041==    by 0x76E011A: ???
==14041==    by 0x76E0580: ???
==14041==    by 0x76E25EB: ???
==14041==    by 0x76E29B4: ???
==14041==    by 0x76E3FDE: ???
==14041==    by 0x74C9C2A: ???
==14041==    by 0x74C8452: ???
==14041==    by 0x74BFC9C: ???
==14041==    by 0x72B37B6: ???
==14041== 
==14041== LEAK SUMMARY:
==14041==    definitely lost: 24,576 bytes in 6 blocks
==14041==    indirectly lost: 0 bytes in 0 blocks
==14041==      possibly lost: 0 bytes in 0 blocks
==14041==    still reachable: 82,764 bytes in 19 blocks
==14041==         suppressed: 0 bytes in 0 blocks
==14041== Reachable blocks (those to which a pointer was found) are not shown.
==14041== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==14041== 
==14041== For counts of detected and suppressed errors, rerun with: -v
==14041== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)

i use libusb-1.0.so  for compile.  do you find this situation?

jadenjiang, 2019-07-04:
hi @msmeissn , i find a memory leak in camera_wait_for_event function that is in camlibs/ptp2/library.c

main code is in below:

			case PTP_CANON_EOS_CHANGES_TYPE_CAMERASTATUS:
					/* if we do capture stuff, camerastatus will turn to 0 when done */
					if (entry.u.status == 0) {
						*eventtype = GP_EVENT_CAPTURE_COMPLETE;
						*eventdata = NULL;
					} else {
						*eventtype = GP_EVENT_UNKNOWN;
						C_MEM (*eventdata = malloc(strlen("Camera Status 123456789012345")+1));
						sprintf (*eventdata, "Camera Status %d", entry.u.status);
					}
					return GP_OK;
				case PTP_CANON_EOS_CHANGES_TYPE_FOCUSINFO:
					*eventtype = GP_EVENT_UNKNOWN;
					C_MEM (*eventdata = malloc(strlen("Focus Info 12345678901234567890123456789")+1));
					sprintf (*eventdata, "Focus Info %s", entry.u.info);

					break;
				case PTP_CANON_EOS_CHANGES_TYPE_FOCUSMASK:
					*eventtype = GP_EVENT_UNKNOWN;
					C_MEM (*eventdata = malloc(strlen("Focus Mask 12345678901234567890123456789")+1));
					sprintf (*eventdata, "Focus Mask %s", entry.u.info);
					break;
				case PTP_CANON_EOS_CHANGES_TYPE_UNKNOWN:
					/* only return if interesting stuff happened */
					if (entry.u.info) {
						*eventtype = GP_EVENT_UNKNOWN;
						*eventdata = entry.u.info; /* take over the allocated string allocation ownership */
						return GP_OK;
					}
					/* continue otherwise */
					break;

-------------------------
when type is PTP_CANON_EOS_CHANGES_TYPE_FOCUSINFO and PTP_CANON_EOS_CHANGES_TYPE_FOCUSMASK, we call malloc function for eventdata pointer and then break. I think we should call  return  replaced break, because when call break, eventdata pointer will get an new memory block in next loop.

msmeissn, i have another question.  is there any different in libusb  and  libusb-1.0. 
when i compile libgphoto2 with-libusb-1.0 and without-libusb,  command  " gphoto2 --auto-detect "  has memory leak, but in reverse i compile libgphoto2 without-libusb-1.0 and with-libusb , that's ok.
do you know why? could you please give me some advice?

msmeissn, 2019-07-07:
you are right with those 2 cases, i added returns.

jadenjiang, 2019-07-11:
Hi @msmeissn 
i find another memory leak when use sony camera that's in ptp mode.
the files will add root folder when sony camera in ptp mode.
so  CameraFilesystem->rootfolder->files that isn't null.  when we call gp_filesystem_free function, we should also free this rootfolder.

--------------------------------------------------------
int  gp_filesystem_free (CameraFilesystem *fs)
{
	/* We don't care for success or failure */
	gp_filesystem_reset (fs);

	/* Now, we've only got left over the root folder. Free that and
	 * the filesystem. */
	free (fs->rootfolder->name);
	free (fs->rootfolder);
    
	free (fs);
	return (GP_OK);
}

**change to:**

int  gp_filesystem_free (CameraFilesystem *fs)
{
	/* We don't care for success or failure */
	gp_filesystem_reset (fs);

	/* Now, we've only got left over the root folder. Free that and
	 * the filesystem. */
	CameraFilesystemFolder  **f = &fs->rootfolder;
	if(*f) delete_folder (fs, f);

	free (fs);
	return (GP_OK);
}

------------------------------------------------------------

do you think right?

msmeissn, 2019-10-12:
this was correct, we did not free the files in the root folder. I adjudsted gp_filesystem_reset instead.

closed 2019-10-12 by msmeissn