gphoto/libgphoto2 #426: Electronic shutter with Nikon Z6 throws -1 on gp_camera_capture

avisecag, 2019-07-03:
**Describe the bug**

I am trying to capture an image with the camera's silent mode (electronic shutter). Related camera widget is _0xD1AD_. I am not able to get the filepath on the camera to grab the image and gp_camera_capture throws error _-1_. This only happens with the silent mode turned on. In the commandline tool I get an unknown error but gphoto2 is still able to download the image from the camera.

**Name the camera**
Nikon Z6

**libgphoto2 and gphoto2 version**

gphoto2         2.5.23
libgphoto2      2.5.23
libgphoto2_port 0.12.0 

**To Reproduce**
Steps to reproduce the behavior:
[electronic_shutter.log -> error](https://github.com/gphoto/libgphoto2/files/3353528/electronic_shutter.log)
[mechanical_shutter.log -> working](https://github.com/gphoto/libgphoto2/files/3353529/mechanical_shutter.log)

avisecag, 2019-07-03:
I was able to resolve the bug:
in file camlibs/ptp2/library.c on line 3520
``C_PTP_REP (nikon_wait_busy (params, 100, 20000))``

must not be executed for Nikon Z 6 and probably Nikon Z 7 (can't test it at the moment)

``	if (params->deviceinfo.Model && (strcmp(params->deviceinfo.Model,"Z 6") != 0)) {
		C_PTP_REP (nikon_wait_busy (params, 100, 20000));
	}``

msmeissn, 2019-07-03:
What happens with this Busy wait code? is it reporting an error and which one?

msmeissn, 2019-07-03:
i see the logfs, sorry ... will check them

msmeissn, 2019-07-03:
0xa201 is the error returned on busy wait

closed 2019-07-06 by msmeissn