gphoto/libgphoto2 #428: Download a picture from the Canon EOS RP using the c api

Patrick-Lascombe, 2019-07-12:
Hello,

I'm looking to download several picture from my canon eos rp camera to my computer (after they were taken) using the c API. 
Is there any example of this scenario ?

Thanks for your help

msmeissn, 2019-07-15:
short snippet:

CameraFilePath path;
CameraFile *file;
char *imagedata;
unsigned long int size;

/* ... init camera, context ... */

gp_file_new (&file);
strcpy(path.folder,"/DCIM/100CANON");
strcpy(path.name,"IMG_012345.JPG");

gp_camera_file_get(camera, path->folder, path->name,GP_FILE_TYPE_NORMAL, file, context);

gp_file_get_data_and_size (file, &imagedata, &size);

now the data is in imagedata and the size of it is in size and can be used further.

msmeissn, 2019-10-12:
no reply - hope my above comment helped.

closed 2019-10-12 by msmeissn