gphoto/libgphoto2 #430: gp_widget_free() frees children, no matter what their ref_count is

ids1024, 2019-07-13:
This may be intended behavior, but I would consider it unexpected (and it isn't mentioned in the documentation).

The obvious solution is to just `gp_widget_deref` the children; this won't work correctly though, since there's a cycle (a widget contains pointers to it's parent and children).

A program using the library, if it is aware of this situation, could address this by just calling `gp_widget_ref` and `gp_widget_deref` on the root of the widget tree. It would also be possible, if desired, for the library to do that automatically when modifying the reference count for a child.

I can send a PR to better document this, but I'd like to see what thoughts there are in what the behavior should be like first.

msmeissn, 2019-07-15:
I think the original author had intended it that way.

it is a bit unsafe, addmitedly, but yes, likely just a doc issue.

ids1024, 2019-07-16:
That's what I thought. It seems like a kind of awkward and confusing API (though most users are probably fine just dealing with the ref count for the root).

It's not really easy to address though.

My thought is to get this behavior (without changing the library) with a wrapper like this:

```c
int ref(CameraWidget *widget) {
    CameraWidget *root;
    int ret = gp_widget_get_root(widget, &root);
    if (ret != GP_OK)
        return ret;
    ret = gp_widget_ref(root);
    return ret;
}

int unref(CameraWidget *widget) {
    CameraWidget *root;
    int ret = gp_widget_get_root(widget, &root);
    if (ret != GP_OK)
        return ret;
    ret = gp_widget_unref(root);
    return ret;
}

```

I think this should be sensible, and avoid both leaks and use after free? I suppose this assumes the parent is not changed after this call.