gphoto/libgphoto2 #434: Manual focus control of Sony A7-III

javierin, 2019-07-31:
**Is your feature request related to a problem? Please describe.**
Is there a way to control manual focus in Sony A7-III camera? I am trying to set up focus stacking using gphoto2 but i can't seem to find the proper key.

**Describe the solution you'd like**
As specified in description

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Sony software is able to control my lenses while in manual focus with 2 levels of normal and fine adjustment. I can provide whatever debug / dumps you need while moving the focus ring if this provides any useful info.

So far, wait-event only displays a handful of 
```
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property d20f changed
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property 5007 changed
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Property d21d changed
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Property d21d changed
```

Which don't provide me any meaningful info related to focusing:
```
gphoto2: {/home/javi} /> get-config /main/other/d21d
Label: PTP Property 0xd21d
Readonly: 0
Type: MENU
Current: 0
Choice: 0 0
Choice: 1 1
Choice: 2 2
END
gphoto2: {/home/javi} /> get-config /main/other/d20f
Label: Color temperature
Readonly: 0
Type: RANGE
Current: 0
Bottom: 2500
Top: 9900
Step: 100
END
gphoto2: {/home/javi} /> get-config /main/other/5004
Label: Compression Setting
Readonly: 0
Type: MENU
Current: 16
Choice: 0 2
Choice: 1 3
Choice: 2 4
Choice: 3 16
Choice: 4 19
Choice: 5 20
Choice: 6 18
END
gphoto2: {/home/javi} /> get-config /main/other/5007
Label: F-Number
Readonly: 0
Type: RANGE
Current: 280
Bottom: 0
Top: 65535
Step: 1
END
```

Thanks

Hutch67, 2019-10-15:
The right command to manually drive the focus is
gphoto2 --set-config=/main/actions/manualfocus=<stepsize>
stepsize is a number from -7 to 7

But there is a bug in libgphoto2, which prevents its function, at least for ILCE-A7M3.
The bug is in file 
libgphoto2-2.5.23/camlibs/ptp2/config.c at line 6819.

There the command 0xd2d2 is sent with value 2 which will enable AutoFocus mode, just befor a manual focus step is sent to the camera.
Just commenting out that line fixes the issue.
So after correcting the file, the lines should look like
`xpropval.u16 = 2;
//C_PTP (ptp_sony_setdevicecontrolvalueb (params, 0xd2d2, &xpropval, PTP_DTC_UINT16));
if(val <= -7) xpropval.u16 = 0xFFFF - 6;
`
Building libgphoto2 and gphoto manually and commenting out the line will bring manualfocus to work.

I did it on ubuntu 19-10 the following way.
Downloaded 
libgphoto2-2.5.23.tar.bz2 
gphoto2-2.5.23.tar.bz2
into my home directory from [https://sourceforge.net/projects/gphoto/files/](url)

To be able to build I hade to install some more components
`sudo apt install autoconf
sudo apt install autopoint
sudo apt install libtool
sudo apt install libusb-dev
sudo apt install pkg-config
sudo apt install libpopt-dev`

Build and install
`tar -xjf libgphoto2-2.5.23.tar.bz2 
tar -xjf gphoto2-2.5.23.tar.bz2
cd libgphoto2-2.5.23/
autoreconf -is
sudo ./configure
sudo make
sudo make install
cd ..
cd gphoto2-2.5.23
autoreconf -is
sudo ./configure
sudo make
sudo make install
`
Then start a fresh terminal and connect your Sony A7M3.
`gphoto2 --set-config=/main/actions/manualfocus=7
gphoto2 --set-config=/main/actions/manualfocus=-7`
Will move the focus back and forth. Lower numbers give smaller steps.

Have fun.

msmeissn, 2019-10-19:
I applied this commenting out ...

I will need to cross check it later on, when I get my hands on a sony that can do that.

closed 2019-10-19 by msmeissn

deanand, 2020-02-11:
I can't seem to get this to work with the Sony A7M4.
Also, is line 8174 supposed to be using d2d1 instead of d2d2? (for the a7m4)

Hutch67, 2020-02-17:
Just sniffed some USB packets of Imaging Edge Remote on Windows 10 controlling a ILCE A7M3
Command with operation code 0x9207 (SONY_SetControlDeviceB) Property 0x0000d2d2 is used to switch the focus mode.
Setting this property to value 0x0001 switches to automatic focus
Setting this property to value 0x0002 switches to manual focus

If in manual focus mode command with operation code 0x9207 (SONY_SetControlDeviceB) Property 0x0000d2d1 is used to move the focus point.
Values between -7 to 7 move the focus in different step sizes back and forth.

Without the "commenting out fix", *every time* before sending a focus step (d2 d1), the focus mode (d2d2) was switched to 0002 (which is manual focus).
Doing this *every time* before a focus step was sent made my A7M3 to block and malfunction.

Hutch67, 2020-02-17:
8174	{ N_("Auto-Focus"),                     "autofocus",        PTP_DPC_SONY_AutoFocus, PTP_VENDOR_SONY,   PTP_DTC_UINT16,  _get_Sony_Autofocus,            _put_Sony_Autofocus },

If looking at 8174 function _put_Sony_Autofocus is used, which sends command ptp_sony_setdevicecontrolvalueb with property PTP_DPC_SONY_AutoFocus and value 2 or 1
PTP_DPC_SONY_AutoFocus is defined as 0xD2C1

So from what I sniffed (see above) I would say 0xD2D2 with a value of 1 should switch to autofocus .
But may be 0xD2C1 works also and may be more generic also with other Sony models.

Hutch67, 2020-02-17:
@deanand
Does manual fucus drive work for A7R4 with Imagine Edge Remote on Windows?
If yes, you may sniff USB traffic generated by Imaging Edge Remote for those commands.
I wrote a small (and experimental) PTP dissector LUA script for wireshark, which helps to find the commands in the stream.
If you want to sniff, record precise times when you are clicking the specific buttons and keep some time delay between pressing the buttons.
E.g. press AF | MF | <<<|<<|<|>|>>|>>>| AF - waiting10 seconds for every | starting at a full minute.

deanand, 2020-02-17:
I've only tried it on OSX and IOS. IOS didn't enable the buttons but OSX did enable the buttons. The buttons don't enable when the lens is set to MF so the lens needs to be in AF plus toggled to Manual mode inside IERemote. I can then control the focus in IERemote once it's in manual mode. I sniffed the PTP/IP over the wifi via wireshark and can see IERemote using 0xD2D2 to set the mode and then 0xD2D1 to step the focus. However this doesn't work in gphoto. I tried setting D2C1 also but none of them worked. Somehow IERemote is setting the mode another way to get it to step focus. I'll try some more sniffing to see if anything else is happening.

Hutch67, 2020-02-17:
@deanand : did you compile libgphoto2 from master branch?
The latest release does not contain the fix.

deanand, 2020-02-19:
I did compile it off the master. Thanks.

ispybadguys, 2021-03-03:
I am wondering if this is all working now in gphoto2. I have tried to change the focus by

gphoto2 --set-config "Near Far=0" 
or 
gphoto2 --set-config "Near Far=7" 

If I executed a get  and I see a MENU of choices but I can't guess the right commands
gphoto2 --get-config "Near Far"
Label: Near Far                                                                
Readonly: 0
Type: MENU
Current: 2
Choice: 0 -7
Choice: 1 -6
Choice: 2 -5
Choice: 3 -4
Choice: 4 -3
Choice: 5 -2
Choice: 6 -1
Choice: 7 0
Choice: 8 1
Choice: 9 2
Choice: 10 3
Choice: 11 4
Choice: 12 5
Choice: 13 6
Choice: 14 7
END

AubinF, 2023-03-21:
Installing libgphoto2 and gphoto2 from source [as suggested by](https://github.com/gphoto/libgphoto2/issues/434#issuecomment-542403104) @Hutch67 then using the following did the trick on my A7-III :

```
# disable AF (use MF)
gphoto2 --set-config /main/capturesettings/focusmode=0

# set focus distance
gphoto2 --set-config /main/actions/manualfocus=7 

# take picture
gphoto2 --capture-image-and-download

# enable AF again
gphoto2 --set-config /main/capturesettings/focusmode=1

``` 

In case this is useful to anyone

- the above works even with AF enabled on my lens.
- I had to set `focusmode` to 0 (MF) for the `manualfocus` command to have any effect.

watsonmw, 2023-03-22:
- the above works only with AF enabled on my lens.

It seems that the AF setting on the lens is really - camera controlled focus.  Since manual focus can only be adjusted via PC Remote / PTP and not via the camera UI, I guess they just named it 'AF' for simplicity.

The 'focus'/half-press mode can get stuck on sometimes as well, causing you not be able change shot settings (e.g. shutter speed, fstop, iso, autofocus mode) in which case you have to power off USB (uhubctl or otherwise disconnect / turn off USB VBUS), and power it back on again.