gphoto/libgphoto2 #438: Sony A7R3 - Download movie

thunders82, 2019-08-18:
**Describe the bug**

I used this command line to record and download a movie :  

gphoto2 --set-config movie=1 --wait-event=10s --set-config movie=0 --wait-event-and-download=5s

While the camera is recording as expected gphoto2 doesn't download the movie on the computer.

**Name the camera**
Sony Alpha-A7R III (Control)

**libgphoto2 and gphoto2 version**
gphoto2 2.5.23.1

**To Reproduce**
Steps to reproduce the behavior:

run the command line : 
gphoto2 --set-config movie=1 --wait-event=10s --set-config movie=0 --wait-event-and-download=5s

I tried to output the log file with the same exact command but because of the additional : 
--debug --debug-logfile=debug.log
gphoto2 doesn't quite after the --wait-event-and-download=5s and requires a double CTRL+C to exit (A bug ?) 
[debug.log](https://github.com/gphoto/libgphoto2/files/3513470/debug.log)

thunders82, 2019-08-20:
Anybody to confirm ? Anything I can test to push this one forward please ?

Thank you

kevings, 2019-08-20:
Not sure if this is helpful but I'm running into this exact problem with the Sony Alpha-A7 III (PC Control), also using versions 2.5.23.1. The debug log worked for me, though. 
[debug.log](https://github.com/gphoto/libgphoto2/files/3522271/debug.log)

msmeissn, 2019-08-21:
On my SOny SLT A58 this is also the case. the Movie is saved to the card but not accessible in the PC Control mode.

thunders82, 2019-08-28:
Anything we can do to help ?

msmeissn, 2019-09-07:
For the older Sony alpha it is a camera side limitation, not a gphoto limitation as far as I understand :(

there is also no "object added" event sent in the traces which  we could act on. :/

thunders82, 2019-10-11:
Hi @msmeissn,
Thank you for looking into it and sorry for my late reply. I was quite busy lately...

Does it mean the recent Sony alpha could be able to download the movies?

msmeissn, 2019-10-20:
I have so far not heard this being possible, not even for newer models ... sorry. /

thunders82, 2019-10-22:
Thank you for the answer. Let's hope Sony will add this feature in the future!

msmeissn, 2021-02-28:
not sure if its possible in newer , someone would need to retest it