gphoto/libgphoto2 #439: Bug with Sony NEX-3N capture image

ccaramel, 2019-08-27:
Hi,
I'm trying to use Sony NEX-3N with INDI and it still not working.
Here is the log file with the latest libgphoto2 obtained with this command line:
`gphoto2 --debug --debug-logfile=gphoto2.txt --capture-image-and-download
`
Any idea?

[gphoto2.txt](https://github.com/gphoto/libgphoto2/files/3547008/gphoto2.txt)

msmeissn, 2019-08-28:
Currently we know only the Sony Alpha series cameras support capture. 
The NEX is not known to support it.
If there are windows or smartphone apps that allow it, we would like to know.

ccaramel, 2019-08-28:
It is worth to notice that the nex-3n is part of the alpha family. 
And it is reported to work on the gphoto/libgphoto page:
http://www.gphoto.org/proj/libgphoto2/support.php

ccaramel, 2019-08-28:
@msmeissn please re-open issue

msmeissn, 2020-03-01:
We currently still assume Sony NEX as not capture capable.

The opcodes for sony capture are not present in the opcode list.