gphoto/libgphoto2 #440: Canon EOS M50 - Err 70

Dlahacz, 2019-08-29:
**Describe the bug**

After x (seems to be random) captured photos err 70 appears on camera screen. Battery have to be removed, to get it working again.

Tested with multiple cameras (all M50) on multiple computers

**Name the camera**
Canon EOS M50

**libgphoto2 and gphoto2 version**
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline 
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF 
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
run this python script for couple of minutes https://gist.github.com/Dlahacz/6e8904ed5786c9c7fb50f5f5f0982377

**debug log**
https://gist.github.com/Dlahacz/c10e4974fb3a7f4abf54edd999a560d0
1-2859 capturing normaly
2860-4598 err 70 appears on camera screen 
4599-4642 gphoto err -110 until batery is removed

msmeissn, 2019-10-05:
error 70 usually means parts of the firmware crashed :/

Dlahacz, 2019-10-29:
Well, now it's crashed even while normal hand capturing. It realy looks like camera problem. I assume you can close the issue. Thanks

bartklaassen, 2019-11-04:
Hi! We're running on various camera's into the same problem. Our M50 is connected to continuous power supply from Canon without any SD card since it is connected via USB. Problem occurs random, already found a solution?

Dlahacz, 2019-11-04:
We have 5 cameras (all M50). 2 of them works without problem. 1 crashes rarely. 2 are not usable at all. I've sent one of them to Canon Service Centre and it came back with note that camera is fine and no repair is needed, but camera is still not working :-1: I'm clueless here.

Dlahacz, 2019-11-04:
s/n 21843047006735 - unusable
s/n 21843047006339 - unusable
s/n 21833047004431- crashes rarely
s/n 21833048*- works fine
s/n 21833049*- works fine

21833047* looks like bad series to me.

bartklaassen, 2019-11-04:
Hi David,

And what do you mean with unusable?
I will check our S/N tomorrow. For what purpose you use the M50?

I’m in touch with Canon about this.

Dlahacz, 2019-11-04:
Unusable means that it crashes all the time (afer ~1hour of usage, it crashes every 1-10 photo)
I use it in same setup as you(no sd and power supply). But it crashes with battery and sd to.

bartklaassen, 2019-11-04:
Yes same here with the crashing. And with the 48*/49* series no problems at
all?

bartklaassen, 2019-11-04:
First camera is also a 47*/ series. The other one I will check.

bartklaassen, 2019-11-05:
Hi,

I’m in touch with a canon tech repair manager (I have a good connection
there). He is willing to look into the problem. Can you describe exactly
how you use it and when it occurs and what you did to test it. Maybe we can
have a call and discuss it and take this problem up to canon together
instead of separate.

Kind regards,
Bart

JStonevalley, 2019-11-20:
Hi,
Has anything happened here? We are currently using one M50 camera (21883049000818) in our setup of ~30 cameras, mixed sony a6000 and canon 750d. The M50 we have seems exposed to this error.
Any news?

Dlahacz, 2019-11-25:
Hi, finally I've received first repaired camera from canon repair center. According to protocol, they replaced some bus system, camera is working fine for now. (and our s/n 21833049* started to crash rarely to)

msmeissn, 2019-11-30:
(leave ticket open ... but it seems there really is some camera side bug/fix going on)

RobertVCampbell, 2020-02-16:
Hi,
I'm not sure if I should be posting here or make my own thread but I'm having the same issue with my Powershot SX740 HS.
I'm trying to capture a sequence of photos with --capture-image-and-download in gphoto2, but the camera frequently crashes with Err 70 appearing onscreen and gphoto showing:

0x2019 PTP Device Busy
Error could not capture image.
Error could not capture.

The crash requires the camera be reset either by the power button or by pulling the battery.

Debug logfile:

bulldoz99, 2020-03-17:
Hi,

I also have same issue(Err70). I also  connected to continuous power supply from Canon without any SD card and  is connected via USB. Some days is okay but other is happening every few minutes. 

Just wondering if anyone find fix for this issue.

Dlahacz, 2020-03-24:
> Hi,
> 
> I also have same issue(Err70). I also connected to continuous power supply from Canon without any SD card and is connected via USB. Some days is okay but other is happening every few minutes.
> 
> Just wondering if anyone find fix for this issue.

Send it to Canon Service Center. It's HW issue. They've already repaired 5 of my devices.

bartklaassen, 2020-03-27:
Hi,

No fix until now, just send it back to Canon repair center under guarantee.
They will replace a part and then it works fine again. Happend to us for 3
out of 4 camera’s (M50) until now.

Kind regards,

Bart

naylors61, 2020-07-15:
Which Canon service centre did you send your camera to. I am in Canada and I would like to be able to advise the service centre  here  that this issue has been resolved by another Canon service centre located where you sent yours. Might save some time.

Dlahacz, 2020-07-16:
> Which Canon service centre did you send your camera to. I am in Canada and I would like to be able to advise the service centre here that this issue has been resolved by another Canon service centre located where you sent yours. Might save some time.

Prague - Czech Republic

naylors61, 2020-09-25:
I see a number of you have had the M50 repair done by canon . My question is has it solved your problem long term and do you know specifically what they did . So what piece did they swap out or repair.

bartklaassen, 2020-09-25:
The problems were solved for us for multiple camera’s, after they were send
to canon service center. But they did not specify what they swapped or
repaired. After that no issues so far.

naylors61, 2020-09-25:
Can you tell me what the location of the Canon repair centre was and if possible a set# from one of the cameras that was repaired. I have some friends at Canon Canada and If I can find out that info they will contact them and see what they did and see if it works on this one.

Thank you

bartklaassen, 2020-09-25:
Please e-mail me about the serial# from the cameras that were repaired, it
was at one of the official canon repair center in the Netherlands.
info@techrepair.nl
You can contact me at bart.klaassen@gmail.com

bartklaassen, 2020-10-01:
Hi,

I checked for you can the PCB ASS'Y, MAIN, W/BATT.
Main PCB was replaced at 3 camera's and worked after that.

naylors61, 2020-10-01:
Thank you I really appreciate the extra effort to find out what was done .

This will be a big help

spiketsi, 2021-01-27:
> Send it to Canon Service Center. It's HW issue. They've already repaired 5 of my devices.

Did you have any issue after canon repair??
I got my m50 fixed from canon and i'm afraid not to have this issue again..

Dlahacz, 2021-01-27:
> Did you have any issue after canon repair??
> I got my m50 fixed from canon and i'm afraid not to have this issue again..

Nope, just other issues like broken shutter etc.

spiketsi, 2021-01-27:
> Nope, just other issues like broken shutter etc.

I'm about to use it for livestreaming about 2-3 hours. Have you tryied to use it for long time recording or using the HDMI output?