gphoto/libgphoto2 #441: Lack of viewfinder widget for olympus OMD camera

joelcollet, 2019-08-30:
Recently owned an OM-D EM5 mkII camera and tried to use it for tethered shooting and liveview with some linux software (like entangle, indilib..etc) for various purpose.
I am struggling with getting the liveview working on all the tools i tried.

Actually in indilib (and also in entangle for what i have been able to decode/understand) there is a call to find a "viewfinder" widget and then set it's value.  However there is no viewfinder widget on these camera and then the software is getting an NOT_SUPPORTED answer which is preventing to get the liveview.
Nevertheless, in indilib , i have been able to modify the code by removing the viewfinder value setting call and going directly to the capture preview, which is working flawlessly !
But on other tools the modification is really hard to do and i am not sure that it is the proper way of doing.
It would be great to have a viewfinder widget in the gphoto library so that the liveview can work flawlessly, even if it is a "fake" one, always returning OK.
Joël

joelcollet, 2020-04-20:
Hello Marcus

sorry for letting  almost 6 monthes passing.
i recenlty discovered on the indilib forum that the same "issue" is also affecting some  Sony mirrorless SLR.
https://www.indilib.org/forum/ccds-dslrs/5881-sony-alpha-5100-viewfinder-widget-error.html#52237.
Can we assume that, since viewfinder is "a mandatory thing" on mirrorles devices , the answer to a call  to viewfinder widget should always return an OK flag / a "fake" widget ?

Joël

msmeissn, 2020-04-26:
The "viewfinder" widget does not always exist... Hmm, I might be able to see if I can add it.

But you can always to preview capture if its supported, even without viewfinder widget :/

msmeissn, 2020-04-26:
indi-3rdparty /  gphoto has meanwhile fiklter code for OLYMPUS and and Sony.

The ability of previews is btw presence of the 
GP_OPERATION_CAPTURE_PREVIEW ability , and not the "viewfinder" widget

joelcollet, 2020-04-27:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px">Indeed i am the one who initially proposed this modification for olympus to the code maintainer so that the liveview can work<br>
(because the viewfinder widget call got NOTSUPPORTED as answer) and this answer was preventing libindi to trigger the liveview)<br>
But maybe this test in the libindi is not the proper one ?<br>
I am also pretty sure that entangle do a call do this viewfinder widget which then prevent execution of the rest of the code when having NOT SUPPORTED answer<br>
but may be we are wrong with the use of this viewfinder widget (which in my understanding correspond to the ability to switch between viewfinder and liveview on a DSLR)..viewfinder that a mirrorless do not have strictly speaking..being always in liveviewing.<br>
<br>
Actually the trick with test on brand in libindi will probably fail when mirrorles devices from "classical SLR makers"  will be used on libindi (eg Nikon, Canon..), or do these canon/nikon mirrorless devices have a viewfinder widget (if /when supported by gphoto)<br>
<br>
Joël</span><br>
<br>
<br>
 
<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Marcus Meissner"<br>
A : "gphoto/libgphoto2" ,"joelcollet" ,"Author"<br>
Envoyé: dimanche 26 Avril 2020 19:51<br>
Objet : Re: [gphoto/libgphoto2] Lack of viewfinder widget for olympus OMD camera (#441)<br>
 
<div class="gl_quoted">
<p> </p>

<p>indi-3rdparty / gphoto has meanwhile fiklter code for OLYMPUS and and Sony.</p>

<p>The ability of previews is btw presence of the<br>
GP_OPERATION_CAPTURE_PREVIEW ability , and not the "viewfinder" widget</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>