gphoto/libgphoto2 #442: Bulb mode stopped working for Nikon D7200

StyXman, 2019-09-04:
**Describe the bug**

Three months ago I connected my computer to my Nikon D7200 via the camera's wifi and took some 180s snapshots without any problems.

Today I tried again (after possibly some `gphoto2` upgrade; I run Debian Sid and I update every Sunday) and the shutter was open for I guess 300s all right, but then something failed and I don't have my shot.

**Name the camera**
`Nikon DSLR (WLAN)`

**libgphoto2 and gphoto2 version**
```
$ gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

**To Reproduce**
Command I used both times:

`LC_ALL=C gphoto2 --camera "Nikon DSLR (WLAN)" --port ptpip:10.84.0.1 --bulb 300 --capture-image-and-download`

I have a `--summary` output from back then when it worked, and I can provide a new one if needed, but please be patient with me, unluckily I don't always have the time. I also have a full debug output form this time, hare are the lines with `failed`:

```
$ grep -i failed my-logfile.txt
0.829250 gp_camera_get_single_config [gphoto2-camera.c:903](0): 'camera->functions->get_single_config ( camera, name, widget, context)' failed: -2
0.853331 _get_config                 (2): Failed to parse value of property 'Lens Name' / 0xd0e0: error code -1
0.947887 _get_config                 (2): Failed to parse value of property 'Auto White Balance Bias' / 0xd017: error code -1
0.949094 _get_config                 (2): Failed to parse value of property 'Tungsten White Balance Bias' / 0xd018: error code -1
0.950544 _get_config                 (2): Failed to parse value of property 'Fluorescent White Balance Bias' / 0xd019: error code -1
0.951703 _get_config                 (2): Failed to parse value of property 'Daylight White Balance Bias' / 0xd01a: error code -1
0.953135 _get_config                 (2): Failed to parse value of property 'Flash White Balance Bias' / 0xd01b: error code -1
0.954322 _get_config                 (2): Failed to parse value of property 'Cloudy White Balance Bias' / 0xd01c: error code -1
0.955758 _get_config                 (2): Failed to parse value of property 'Shady White Balance Bias' / 0xd01d: error code -1
1.487879 ptp_usb_getresp [usb.c:465] (0): PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
1.588597 ptp_usb_getresp [usb.c:465] (0): PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
1.689469 ptp_usb_getresp [usb.c:465] (0): PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
3.957985 ptp_usb_getresp [usb.c:465] (0): PTP_OC 0x100b receiving resp failed: PTP Invalid Object Handle (0x2009)
3.958011 camera_nikon_capture [library.c:3100](0): deleteobject(2b000001) failed: 2009
3.958017 camera_nikon_capture [library.c:3106](0): deleteobjectinsdram(2b000001) failed too: 2009
0.025576 gp_port_info_list_lookup_path(2): re_match failed (-1)
300.976872 _put_Nikon_Bulb [config.c:6736](0): 'nikon_wait_busy(params, 100, 5000)' failed: (null) (0xa200)
300.976891 gp_camera_set_single_config [gphoto2-camera.c:1145](0): 'camera->functions->set_single_config ( camera, name, widget, context)' failed: -1
300.976912 gp_context_error            (0): Failed to set new configuration value 0 for configuration entry bulb.
```

I will try to upload the full outputs and debug soon.

StyXman, 2019-09-04:
I have another shot from when it worked and it was for 299s. Maybe exactly 300s (or more) is not supported?

msmeissn, 2019-10-12:
0xa200 is a camera error.... The camera might not like this long time I think.

msmeissn, 2019-10-12:
we do not have a 300 second limit at least (or any limit in this case)

StyXman, 2019-10-17:
What would yo need for pushing this further?

msmeissn, 2019-10-20:
a full debuginfo of this would be helpful

StyXman, 2021-12-25:
Dang, 2y later and I still get this. I just tried again. 

This works:

```
mdione@ioniq:~/src/projects/ansel$ gphoto2 --set-config f-number=f/10 --capture-image-and-download 
New file is in location /store_00010001/DCIM/117D7200/NKN_6801.JPG on the camera
Saving file as NKN_6801.JPG
Deleting file /store_00010001/DCIM/117D7200/NKN_6801.JPG on the camera
```

but this doesn't:

```
mdione@ioniq:~/src/projects/ansel$ gphoto2 --set-config bulb=1 --wait-event 2s --capture-image-and-download 
Waiting for 40 seconds for events from camera. Press Ctrl-C to abort.          
UNKNOWN PTP Property 500e changed
UNKNOWN PTP Property 500e changed
UNKNOWN PTP Property 5013 changed
UNKNOWN PTP Property d1b3 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1f1 changed
UNKNOWN PTP Property 500c changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed

*** Error ***              
PTP Device Busy
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***       
```

Attached [full debug file](https://github.com/gphoto/libgphoto2/files/7775401/failed_bulb.log).

StyXman, 2021-12-25:
I just tried the same command and I noticed that at least this time, the photographs stay in the camera's SD. I have the impression it's not always the case.