gphoto/libgphoto2 #443: Setting datetime incorrectly handles timezones

wtgee, 2019-09-09:
**Describe the bug**
I've been having issues setting the time correctly (see https://github.com/panoptes/POCS/pull/892) and realized it seems to be flipping a negative sign on the timezone when setting. Steps to reproduce below.

**Name the camera**
```
➜ gphoto2 --auto-detect                           
Model                          Port                                            
----------------------------------------------------------
Canon EOS 100D                 usb:001,008     
Canon EOS 100D                 usb:001,009   
```

**libgphoto2 and gphoto2 version**
```
➜ gphoto2 --version
gphoto2 2.5.23.1

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.12         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking
```

**To Reproduce**
I can get the debug log but it doesn't show anything useful.

```
➜ gphoto2 --port usb:001,009 --set-config datetime=now       

➜ gphoto2 --port usb:001,009 --get-config datetime    
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567988058
Printable: Mon 09 Sep 2019 10:14:18 AEST
Help: Use 'now' as the current time when setting.

END

➜ date
Mon  9 Sep 10:14:22 AEST 2019

➜ gphoto2 --port usb:001,009 --get-config datetimeutc
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1568060066
Printable: Tue 10 Sep 2019 06:14:26 AEST
Help: Use 'now' as the current time when setting.

END
```

Note that it has taken the `datetime` and added 20 hours to it. If it thought `datetime` was UTC and added the regular +10 (which is my GMT offset) then added another +10.

```
➜ gphoto2 --port usb:001,009 --set-config datetimeutc=now

➜ gphoto2 --port usb:001,009 --get-config datetimeutc    
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567988099
Printable: Mon 09 Sep 2019 10:14:59 AEST
Help: Use 'now' as the current time when setting.

END

➜ gphoto2 --port usb:001,009 --get-config datetime   
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567916103
Printable: Sun 08 Sep 2019 14:15:03 AEST
Help: Use 'now' as the current time when setting.

END
```
If I set `datetimeutc` with `now` it then seems to do -20.

I can try this with some other cameras I have in different timezones as well, will do and report back.

wtgee, 2019-09-09:
```
➜ gphoto2 --version
gphoto2 2.5.17

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.17         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.18         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

➜ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 100D                 usb:002,030     
Canon EOS 100D                 usb:002,029     

➜  date
Sun Sep  8 14:27:01 HST 2019

➜ gphoto2 --port usb:002,030 --get-config datetime
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1568024833
Printable: Mon 09 Sep 2019 12:27:13 AM HST
Help: Use 'now' as the current time when setting.

END

➜ gphoto2 --port usb:002,030 --get-config datetimeutc 
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567988844
Printable: Sun 08 Sep 2019 02:27:24 PM HST
Help: Use 'now' as the current time when setting.

END
```

Trying to set datetime doesn't even work here but I can't do an update right now.

```
➜ gphoto2 --port usb:002,030 --set-config datetime=now

➜ gphoto2 --port usb:002,030 --get-config datetime    
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567988928
Printable: Sun 08 Sep 2019 02:28:48 PM HST
Help: Use 'now' as the current time when setting.

END

➜ gphoto2 --port usb:002,030 --set-config datetimeutc=now

➜ gphoto2 --port usb:002,030 --get-config datetimeutc    
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1567988998
Printable: Sun 08 Sep 2019 02:29:58 PM HST
Help: Use 'now' as the current time when setting.

END

➜ gphoto2 --port usb:002,030 --get-config datetime       
Label: Camera Date and Time                                                    
Readonly: 0
Type: DATE
Current: 1568025011
Printable: Mon 09 Sep 2019 12:30:11 AM HST
Help: Use 'now' as the current time when setting.

END

➜ date
Sun Sep  8 14:30:19 HST 2019

```

wtgee, 2019-09-09:
I'm trying to get the `camera_summary`  info to show somewhere but can't get it to show. 

https://github.com/gphoto/libgphoto2/blob/d8fda3263c35da576fbcbf15e6c287204cefdbe9/camlibs/canon/library.c#L1053-L1061

I'm using:

```
gphoto2 --port usb:001,009 --summary --debug --debug-loglevel=debug --debug-logfile=gphoto2.log
```

msmeissn, 2019-09-15:
for EOS try as workaround to use datetimeutc   which avoids the conversion issues

msmeissn, 2019-09-15:
hmm, i mean is datetimeutc working for you? I need to review this code again :/

wtgee, 2019-09-17:
Yes, as indicated neither of them seem to work quite as expected. Will try to do some more debugging within the code when I get some time but any pointers to appropriate sections of code would be great.

Thanks!

cClaude, 2021-06-04:
Issue exist with `Canon EOS 700D` and `gphoto2 2.5.23`

```
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

```
Camera summary:
Manufacturer: Canon Inc.
Model: Canon EOS 700D
  Version: 3-1.1.4
  Serial Number: REMOVED
Vendor Extension ID: 0xb (1.0)
```

ndim, 2021-06-04:
Are camera still time zone agnostic these days, or have they perhaps started to do time zone corrections on their own?

cClaude, 2021-06-05:
I'm not sure I would qualify this problem as it's done here.

There is a problem with the camera (CANON 700D in my case). It manages the time zones but not the time changes between summer and winter.

My zone is Paris and for the camera, it's +1 year round compared to UTC, whereas in reality it's +1 in winter and +2 in summer.

My workaround is as follows:

1. Configure the camera for London (UTC +0)
2. Then I use gphoto2 as follows:

```bash
gphoto2 --set-config /main/settings/datetimeutc=$( date -u +"%s" --date="$( date +'%F %T' )" )
gphoto2 --get-config datetime
```

My observation is that having partial time zone support is worse than not having any at all.

I therefore deactivate the time zone support by declaring being in the UTC zone.

Then I convert the calculation of the current time to seconds regardless of the time zone.

wtgee, 2021-06-08:
Thanks for the updates from others. I haven't had a chance to look into this further. My workaround is to just put a timestamp in the filename, which comes from the OS itself. I then ignore what goes into the EXIF. This is, of course, not ideal.

wtgee, 2021-06-09:
> a better workaround is to just use UTC.

As shown above, the problem is that `datetimeutc=now` is not in fact setting the datetime to UTC, which is what I was trying to do.

@cClaude gets around this by explicitly setting the date rather than using the currently documented method. 

However my cameras are located remotely so I don't have physical access to them and have no ability to change the timezone or daylight savings settings as those are not available via gphoto2. These are set to my current timezone and therefore if I specifically set the date as per @cClaude I still get the same kind of results as shown in my original posts.

So for me the better workaround is to continue to set the timestamp in the filename.