gphoto/libgphoto2 #444: /usr/include/gphoto2/gphoto2 infinite loop

hwpfeil, 2019-09-14:
**Describe the bug**

libgphoto2 installs /usr/include/gphoto2/gphoto2 which is a link pointing to './'

A clear and concise description of what the bug is.

Infinite recursion. gphoto2-->gphoto2-->gphoto2....
Is there a compelling reason for this self-recursion?

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
n/a
**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

2.5.23

**To Reproduce**
Steps to reproduce the behavior:

ls -l /usr/include/gphoto2/gphoto2
ls -l /usr/include/gphoto2/gphoto2/gphoto2
ls -l /usr/include/gphoto2/gphoto2/gphoto2/gphoto2
...

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

ndim, 2019-09-18:
This was introduced for source compatibility between the two forms

    #include <gphoto2.h>
    #include <gphoto2/gphoto2.h>

Of the top of my head, I cannot remember the exact scenario where this was introduced, I am looking at "git blame" and `git show fd692a071e` right now to help me remember.

What exactly is the problem with that symlink? Which program is exhibiting that "infinite recursion" behaviour?

ndim, 2019-09-18:
In September 2006, I made

     #include <gphoto2/gphoto2-foo.h>

the consistent way to include `libgphoto2_port` and `libgphoto2` headers (cf. https://github.com/gphoto/libgphoto2/commit/b8687e532ff267edd641064884e1cd506621fd1b). For compatibility with third party software using `libgphoto2` and still using `#include <gphoto2-foo.h>`, I added that symlink.

I just found that the Fedora package has been removing the symlink for their `libgphoto2` package since 2009 ([RHBZ #460807](https://bugzilla.redhat.com/show_bug.cgi?id=460807)) so there is probably no software around any more now (ten years later, in September 2019) which requires this.

So `libgphoto2` (and `libgphoto2_port`) could maybe drop the symlink at this time.

I still think if some program recursively traversing the filesystem tree (such as e.g. `grep -r` or `find`) can be confused by symlinks pointing somewhere, it is those programs which should be fixed. This is a different problem, though.

hwpfeil, 2019-09-18:
I think it indeed may have been a "grep -r" or "find" that found the strange directory. I was probably looking for an include file that I needed to add.  I don't think it is an issue that needs to be addressed.  Just something strange that I happened to notice. If it works, don't fix it.

ndim, 2019-09-18:
I would like to hear @msmeissn on this. I have not checked Suse, Debian, and other distros (yet), but if Fedora has been living without this symlink for the past ten years, this symlink appears to be pretty useless these days.

It would remove some build code and strange installation artefacts from libgphoto2, which is good.

And if it saves some people with broken symlink handling in their `grep -r` or `find` or similar recursive descent filesystem walk algorithms from infinite recursion, that would be a welcome side effect.

hwpfeil, 2019-09-19:
Aside: I vaguely recall that the hunt, whether grep or find -exec grep, just noticed the self-reference. The search didn't follow the link, no recursive descent. In response to the unexpected message indicating the self-reference, I checked the path with ls -l. (ext4 filesystem) The link points to './' which is the entire directory. Each iteration contains another link to './'. 
Thank you for gphoto2!

closed 2019-10-18 by msmeissn