gphoto/libgphoto2 #448: Pentax K30 shutter does not fire.

lkiersey, 2019-10-06:
I have tried using my Pentax K30 on the Linux KSTARS app for astrophotography which runs on the libgphoto2 driver. The camera is detected automatically and knows that it is a K30, it is also able to set the ISO correctly when I connect. However, when I try to get it to fire the shutter it times out and fails the exposure. This happens no matter what exposure setting I request or whether I am in bulb mode or not.
If I manually press the shutter during the period where it is trying to expose it downloads that exposure and reports a successful exposure. So the issue appears to be just around actually triggering the shutter release (I have tried it in both bulb mode settings i.e. fire once to start and a second to finish or the hold to expose)
I know there have been issues with this on the K30/K50 series but I was wondering is there a solution to this issue?

msmeissn, 2019-10-12:
I have not much knowledge of the Pentax driver I am afraid, but can you capture a debuglogfile with the commandline gphoto2 tool?

gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log

lkiersey, 2019-10-13:
[k30_bulb_attempt.txt](https://github.com/gphoto/libgphoto2/files/3721263/k30_bulb_attempt.txt)

So I was able to capture an image but when i try to capture an image in bulb mode i run into errors, see attached log files.
Bulb mode setting on camera was tried at 1 and 2 
[succesful_capture.zip](https://github.com/gphoto/libgphoto2/files/3721265/succesful_capture.zip)