gphoto/libgphoto2 #449: No error raised when camera is getting shutdown, Canon RP

Patrick-Lascombe, 2019-10-07:
Hello,

I'm using a Canon RP camera, and I'm trying to detect through the c library when the camera is turned off.
I used gp_context_set_error_func to recover all the error raised. It works very well for a lot of errors, but I don't receive anything when I turn off the camera. Is that normal ? Is there no error raised when the connection with the camera is lost ?

Thanks for your help !

msmeissn, 2019-10-07:
you will only get a callback if we see an IO error.

usually called api functions will start returning IO Errors.

Patrick-Lascombe, 2019-10-07:
Thank you for your answer, your solution worked but since we didn't wanted to send request to the camera that might add a delay to the picture request, we checked if the usb is active in a separate thread.
Works like a charm

closed 2019-10-07 by Patrick-Lascombe