gphoto/libgphoto2 #450: Problem download with Nikon D7100 Mac OS X

apsylone, 2019-10-12:
Hello

I've tried a simple thing : capturing movie with the command provided by the documentation : 
`env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config movie=1 --wait-event=10s --set-config movie=0 --wait-event-and-download=2s`

My camera is a Nikon DSC D7100. My environnement is Mac OS Mojave Version 10.14.5

`gphoto2 --version
gphoto2         2.5.23         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, clang, ltdl, USB, serial without locking`

Log file is attached.

Could you help me with this problem ?

Thanks for your help,

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/3720714/my-logfile.txt)

msmeissn, 2019-10-12:
It seems to work, the new file is reported even ... but the directory read times out after 20 seconds.

how many files are on the camera?

is gphoto2 -L   working?

apsylone, 2019-10-12:
There are some files recorded, the Nikon has successfluy done the movie. But It don't download it.

msmeissn, 2019-10-20:
Can you add a -L in front:

env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt -L --set-config movie=1 --wait-event=10s --set-config movie=0 --wait-event-and-download=2s

As the directory retrieval after movie capture seems to time out we can try to do it before.