gphoto/libgphoto2 #451: Feature: Skip broken files

angry-elf, 2019-10-17:
I'm using gphoto2 to download files (for backup) from my iPhone7:

```
gphoto2 -P --skip-existing
```

I stuck with the problem - sometimes, gphoto stuck on some type of errors, linked with some erroneous images. 

Debug log show errors like:
```
Transfer 0x55b7fd5c54b0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3
```

After investigation, I definetelly can say, issue is linked with some types of images. If I skip this images (by providing range of image ID to sync), gphoto can continue without problems. 

I'm using gphoto-fs FUSE wrapper to avoid broken files, but it's better to add a feature to gphoto tools to skip this broken files and continue syncing.

Is it possible?

msmeissn, 2019-10-18:
the transfer error is something different.
question is what the getobject returns for those

Is the communication recovering after such a broken object or not?

Can you attach a debuglog of downloading 1 broken object specifically?

angry-elf, 2019-10-18:
> the transfer error is something different.
> question is what the getobject returns for those
> 
> Is the communication recovering after such a broken object or not?
> 
> Can you attach a debuglog of downloading 1 broken object specifically?

I've submitted new issue as bug report with log files - https://github.com/gphoto/libgphoto2/issues/453