gphoto/libgphoto2 #453: Syncing iphone cause multiple transfer errors

angry-elf, 2019-10-18:
I'm using gphoto cmdline tool to sync photos from my iPhone7 via USB.

Sometimes, on specified files, I stuck with the error. It happens on first try (like listing files) and on syncing files, on specified file. Seems like some pictures are broken and gphoto can't get them in any way in any try (pictures are ok, when viewing from iphone app, or via photos application in mac os x). 

So, the bug can occure when I'm listing files (```gphoto2 -l```)  and syncing files (```gphoto2 --skip-existing -P```).

Camera name:
```
$ LANG=C gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Apple iPhone 5 (PTP mode)      usb:001,013     
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
```
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         x86_64-pc-linux-gnu-gcc, popt(m), exif, no cdk, no aa, no jpeg, readline
libgphoto2      2.5.23         INCOMPLETE CAMLIB SET (ptp2), x86_64-pc-linux-gnu-gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip usb1 usbdiskdirect usbscsi, x86_64-pc-linux-gnu-gcc, ltdl, USB, no serial
```

**To Reproduce**
Steps to reproduce the behavior:
```
LANG=C gphoto2 -l --debug --debug-logfile=list.log
There is 1 folder in folder '/'.                                               
 - store_00010001

*** Error ***              
PTP Invalid Storage ID
*** Error (-1: 'Unspecified error') ***       
```

[list.log](https://github.com/gphoto/libgphoto2/files/3742962/list.log)

msmeissn, 2019-10-19:
For this logfile: This error happens on unlocked devices (we do not receive the unlocked storage id, it is still 0xfeedface).

can you get a trace of the aborting download?

angry-elf, 2019-10-21:
Hi.
Just tried transfer some files. Two files (1323 - JPG and 1326 - MOV) transferred ok, third file (1324 - MOV) raises error -7.
Logs are too big, I've uploaded them to my host:

http://elfov.net/tmp/gphoto/file1323-good.log
http://elfov.net/tmp/gphoto/file1324-error-7.log
http://elfov.net/tmp/gphoto/file1326-good.log

Algebro7, 2019-11-13:
I believe I may be running into the same issue on my iphone7+:

```
Nov 13 16:20:44 arch digikam[98090]: digikam.import: Libgphoto2 error:  I/O problem  ( -7 )
Nov 13 16:20:44 arch digikam[98090]: digikam.import: Log ( "IMG_2729.MOV"   "/store_00010001/DCIM/102APPLE/" :  "Failed to download ‘IMG_2729.MOV’"
Nov 13 16:20:44 arch digikam[98090]: digikam.import: Downloading:  "IMG_2730.JPG"  using  "/home/algebro/Pictures/iphone/2019-08/Camera-tmp1-98090.digikamtempfile.IMG_2730.JPG"
Nov 13 16:20:44 arch digikam[98090]: digikam.import: Failed to get camera item!
```

Running Arch Linux with libgphoto2 2.5.23. It happens when I try to import my library with digikam--it will run for a few minutes and then suddenly every transfer will fail with this error. I have to close digikam, unplug the device and plug it back in to get it to work again.

krzyk, 2021-07-25:
What worked for me (at least 6 months ago) was to disable automatic conversion of photos from heic to jpg (there is such option in settings on iPhone). But now it does not work and I still geti I/O error.