gphoto/libgphoto2 #454: Ricoh Caplio G4 - import error

herrdeh, 2019-10-20:
**Describe the bug**
Yes, I know, this camera is *very* old. But I'm used to it, and it does a good job. So if someone is really passionate, he'd do me a great favour to fix it.

A clear and concise description of what the bug is.
Camera is no longer recognized. It used to work till the moment I tried to upgrade libgphoto2 to 

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
Ricoh Caplio G4
USB PTP Class Camera           usb:001,018
**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

gphoto2 2.5.15
Diese Version von gphoto2 benutzt die folgenden Softwareversionen und Optionen:
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.23.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:
Plug in the camera.
When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/3748090/my-logfile.txt)

msmeissn, 2019-10-20:
In 2.5.23 I started no longer building outdated camera drivers by default ... they can still be enabled:

        ./configure --with-camlibs=all,outdated

As you are still a user, I will reenable it in 2.5.24 :)

closed 2019-10-20 by msmeissn

herrdeh, 2019-10-20:
Sounds great ! ((((((:+1: 
Is there a way to enable previews as well? - They never worked...

msmeissn, 2019-10-21:
I dont know... :/ Did the windows software allow it?

herrdeh, 2019-10-21:
What a question..... (-;
I NEVER EVER used a windows machine. Could one ever?

hfiguiere, 2019-10-21:
Maybe it is written in the fine manual that came with the camera?

herrdeh, 2019-10-21:
Am Montag, den 21.10.2019, 08:30 -0700 schrieb Hubert Figuière:
> Maybe it is written in the fine manual that came with the camera?

Actually, I found a PDF manual of the camera in my archive. B-)
It says that you can see miniature previews using a windows software
called DU-10x. Whatever that means....

msmeissn, 2019-10-26:
I think this refers to thumbnails for images.

The camera is from a time where liveview was not yet common in inexpensive models, so I would say it is not supported.

herrdeh, 2019-11-07:
Yes, this is sort of extreme sports in computing. But who knows, maybe someone likes it.... (-;

I downgraded to 
```
ii  libgphoto2-6:amd64       2.5.16-2          amd64             gphoto2 digital camera library
ii  libgphoto2-l10n          2.5.16-2          all               gphoto2 digital camera library - localized messages
un  libgphoto2-port10        <keine>           <keine>           (keine Beschreibung vorhanden)
ii  libgphoto2-port12:amd64  2.5.16-2          amd64             gphoto2 digital camera port library
```
in the meanwhile. Such, I restored the status before: 
I can see the Ricoh G4 at my file browser (nautilus) and import photos. Slow, but working. 

What I cannot do:
Import photos to any photo programme. Neither shotwell nor digikam will do. 

Here is what I can find at terminal with digikam:
```
$ digikam 
digikam.general: AlbumWatch use QFileSystemWatcher
digikam.general: Database Parameters:
   Type:                     "QSQLITE"
   DB Core Name:             "/home/wd/Bilder/digikam4.db"
   DB Thumbs Name:           "/home/wd/Bilder/thumbnails-digikam.db"
   DB Face Name:             "/home/wd/Bilder/recognition.db"
   Connect Options:          ""
   Host Name:                ""
   Host port:                -1
   Internal Server:          false
   Internal Server Path:     ""
   Internal Server Serv Cmd: ""
   Internal Server Init Cmd: ""
   Username:                 ""
   Password:                 ""

digikam.dbengine: Loading SQL code from config file "/usr/share/digikam/database/dbconfig.xml"
digikam.dbengine: Checking XML version ID => expected:  3  found:  3
digikam.coredb: Core database: running schema update
digikam.coredb: Core database: have a structure version  8
digikam.coredb: Core database: makeUpdates  8  to  8
digikam.database: Creating new Location  "/wd/Bilder"  uuid  "volumeid:?uuid=c1317d32-4f15-484f-bbe8-1bb14a6ad335"
digikam.database: Creating new Location  "/wd/Dropbox/photos_vom_FuT"  uuid  "volumeid:?uuid=c1317d32-4f15-484f-bbe8-1bb14a6ad335"
digikam.database: location for  "/home/wd/Bilder"  is available  true
digikam.database: location for  "/home/wd/Dropbox/photos_vom_FuT"  is available  true
KMemoryInfo: Platform identified :  "LINUX"
KMemoryInfo: TotalRam:  12524572672
digikam.general: Allowing a cache size of 200 MB
digikam.thumbsdb: ThumbDB SelectThumbnailSetting val ret =  0
digikam.thumbsdb: ThumbDB SelectThumbnailSetting val ret =  0
digikam.thumbsdb: Thumbs database: have a structure version  "3"
digikam.general: Thumbnails database ready for use
digikam.dimg: ("/usr/share/color/icc", "/home/wd/.local/share/icc")
digikam.dimg: No X.org XICC profile installed for screen  1
digikam.general: Camera XML data:  "/home/wd/.local/share/digikam/cameras.xml"
digikam.facedb: FaceDB SelectFaceSetting val ret =  0
digikam.facedb: FaceDB SelectFaceSetting val ret =  0
digikam.facedb: Face database: have a structure version  "2"
digikam.facesengine: Face database ready for use
digikam.general: Face PipeLine: add database writer
digikam.general: Face PipeLine: add faces trainer
digikam.facesengine: Face database ready for use
digikam.general: Face PipeLine: add database writer
digikam.general: Face PipeLine: add faces trainer
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
digikam.geoiface: "setting backend marble"
QtAV 1.12.0(Apr 15 2018, 02:52:19)
Multimedia framework base on Qt and FFmpeg.
Distributed under the terms of LGPLv2.1 or later.
Shanghai University->S3 Graphics->Deepin->PPTV, Shanghai, China Copyright (C) 2012-2016 Wang Bin (aka. Lucas Wang) wbsecg1@gmail.com
Donate: http://qtav.org/donate.html
[... 187 lines cut ...]
UdevQt: unhandled device action "bind"
UdevQt: unhandled device action "bind"
UdevQt: unhandled device action "unbind"
UdevQt: unhandled device action "bind"
UdevQt: unhandled device action "unbind"
UdevQt: unhandled device action "bind"
UdevQt: unhandled device action "unbind"
UdevQt: unhandled device action "bind"
digikam.geoiface: "setting backend marble"
digikam.general: AudioOutput backends: ("OpenAL", "Pulse", "null")
digikam.geoiface: "ROADMAP"
digikam.geoiface: "setting backend marble"
QLayout: Attempting to add QLayout "" to Digikam::DNGConvertSettings "", which already has a layout
digikam.geoiface: "ROADMAP"
QFSFileEngine::open: No file name specified
digikam.import: Log ( ""   "" :  "Verbindung mit der Kamera wird hergestellt ..."
digikam.geoiface: ----
digikam.import: Failed to initialize camera!
digikam.import: Libgphoto2 error:  Could not claim the USB device  ( -53 )
digikam.import: Log ( ""   "" :  "Verbindung fehlgeschlagen."
QWidget::insertAction: Attempt to insert null action
QFSFileEngine::open: No file name specified
(...)
```
Is there anything I can do?

msmeissn, 2019-11-09:
digikam.import: Libgphoto2 error:  Could not claim the USB device  ( -53 )

This can be a permission error, or that means gvfs or another process is accessing the camera already. As you Nautilus works, it is probably gvfs.

you need to kill / disable the gvfs gphoto2 slaves

hfiguiere, 2019-11-09:
Unmounting in Nautilus it ought to be enough.

herrdeh, 2019-11-10:
Great you're trying to deal with this really old stuff.... (-:
I unmounted the camera in nautilus, then launched digikam and tried to import. Actually, I can see the files in the import dialogue, but no previews (as expected). When I try to import images, programme works, but I end up with files of 0 byte.  Importing these files with the SC card reader still is no problem.

This is terminal output:
```
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014897.JPG: The file contains data of an unknown image type
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> Title:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> Comment:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> Pick Label:  0
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> Color Label:  0
digikam.metaengine: Rating value to write is out of range!
digikam.general: Writting tags
digikam.general: Delete all keywords
digikam.metaengine: xmlACDSee "<Categories></Categories>"
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> Read Iptc Keywords:  ()
digikam.metaengine: "/home/wd/Bilder/R0014897.JPG"  ==> New Iptc Keywords:  ()
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014897.JPG"
digikam.metaengine: Cannot save metadata to image using Exiv2   (Error # 11 :  /home/wd/Bilder/R0014897.JPG: The file contains data of an unknown image type
digikam.metaengine: Metadata for file "R0014897.JPG" written to file.
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014897.JPG: The file contains data of an unknown image type
digikam.metaengine: Exif.Image.Orientation tag set to:  1
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014897.JPG"
digikam.metaengine: Cannot save metadata to image using Exiv2   (Error # 11 :  /home/wd/Bilder/R0014897.JPG: The file contains data of an unknown image type
digikam.metaengine: Metadata for file "R0014897.JPG" written to file.
digikam.general: No write to baloo +++++++++++++++++++++++++++++++++++++
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014900.JPG: The file contains data of an unknown image type
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> Title:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> Comment:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> Pick Label:  0
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> Color Label:  0
digikam.metaengine: Rating value to write is out of range!
digikam.general: Writting tags
digikam.general: Delete all keywords
digikam.metaengine: xmlACDSee "<Categories></Categories>"
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> Read Iptc Keywords:  ()
digikam.metaengine: "/home/wd/Bilder/R0014900.JPG"  ==> New Iptc Keywords:  ()
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014900.JPG"
digikam.metaengine: Cannot save metadata to image using Exiv2   (Error # 11 :  /home/wd/Bilder/R0014900.JPG: The file contains data of an unknown image type
digikam.metaengine: Metadata for file "R0014900.JPG" written to file.
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014900.JPG: The file contains data of an unknown image type
digikam.metaengine: Exif.Image.Orientation tag set to:  1
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014900.JPG"
digikam.metaengine: Cannot save metadata to image using Exiv2   (Error # 11 :  /home/wd/Bilder/R0014900.JPG: The file contains data of an unknown image type
digikam.metaengine: Metadata for file "R0014900.JPG" written to file.
digikam.general: No write to baloo +++++++++++++++++++++++++++++++++++++
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014898.JPG: The file contains data of an unknown image type
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> Title:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> Comment:  QMap()
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> Pick Label:  0
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> Color Label:  0
digikam.metaengine: Rating value to write is out of range!
digikam.general: Writting tags
digikam.general: Delete all keywords
digikam.metaengine: xmlACDSee "<Categories></Categories>"
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> Read Iptc Keywords:  ()
digikam.metaengine: "/home/wd/Bilder/R0014898.JPG"  ==> New Iptc Keywords:  ()
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014898.JPG"
digikam.metaengine: Cannot save metadata to image using Exiv2   (Error # 11 :  /home/wd/Bilder/R0014898.JPG: The file contains data of an unknown image type
digikam.metaengine: Metadata for file "R0014898.JPG" written to file.
digikam.metaengine: Cannot load metadata from file   (Error # 11 :  /home/wd/Bilder/R0014898.JPG: The file contains data of an unknown image type
digikam.metaengine: Exif.Image.Orientation tag set to:  1
digikam.metaengine: MetaEngine::metadataWritingMode 0
digikam.metaengine: Will write Metadata to file "/home/wd/Bilder/R0014898.JPG"
[... cut ...]
digikam.dimg: "/home/wd/Bilder/R0014900.JPG"  : QIMAGE file identified
digikam.dimg.qimage: Can not load " "/home/wd/Bilder/R0014900.JPG" " using DImg::QImageLoader!
digikam.general: Cannot create thumbnail for  "/home/wd/Bilder/R0014900.JPG"
digikam.general: Thumbnail is null for  "/home/wd/Bilder/R0014900.JPG"
digikam.general: Request to get thumbnail for  "/home/wd/Bilder/R0014900.JPG"
digikam.general: Mime type is not video or audio/mpeg from  "R0014900.JPG"
digikam.general: Failed to extract video thumbnail for  "/home/wd/Bilder/R0014900.JPG"
```