gphoto/libgphoto2 #455: (Canon EOS R): inconsistent implementation of shutterspeed and aperture commands

rpg777, 2019-10-25:
**Describe the bug**
`--set-config /main/capturesettings/shutterspeed` and `--set-config /main/capturesettings/aperture` both yield different settings on the camera depending on which option selected. Sometimes the setting code is properly interpreted, sometimes the setting code is interpreted as a string literal. 

**NOTE:** 
**Aperture has similar issues, and gphoto2 always reports ISO to be in "Auto" mode even when explicitly set.** 

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

```
0.099419 print_debug_deviceinfo      (2): Device info:
0.099440 print_debug_deviceinfo      (2): Manufacturer: Canon.Inc
0.099453 print_debug_deviceinfo      (2):   Model: Canon EOS R
0.099466 print_debug_deviceinfo      (2):   device version: 3-1.2.0
```

**libgphoto2 and gphoto2 version**
```
$ gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, clang, ltdl, USB, serial without locking
```

**To Reproduce**
Steps to reproduce the behavior:

`gphoto2 --set-config /main/capturesettings/shutterspeed=25`
expected result: 1/10" shutter
actual result: 25" shutter

`gphoto2 --set-config /main/capturesettings/shutterspeed=26`
expected result: 1/13" shutter
actual result: 1/13" shutter (correctly interpreted setting code)

Also attaching full debug output from two attempts, one with a successful result ([ss31.txt](https://github.com/gphoto/libgphoto2/files/3773748/ss31.txt)), and one with this literal interpretation behavior ([ss30.txt](https://github.com/gphoto/libgphoto2/files/3773747/ss30.txt)).

Also my rough notes on each shutterspeed code and how they are interpreted by gphoto2:
```
Choice: 0 30        shutterspeed=0  30"     OK      shutterspeed=30 30" 
Choice: 1 25        shutterspeed=1  (1")    ERROR*  shutterspeed=25 25"
Choice: 2 20        shutterspeed=2  (2")    ERROR*  shutterspeed=20 20"
Choice: 3 15        shutterspeed=3  15"     OK      shutterspeed=15 15"
Choice: 4 13        shutterspeed=4  (4")    ERROR*  shutterspeed=13 13"
Choice: 5 10        shutterspeed=5  (5")    ERROR*  "10" will work
Choice: 6 8         shutterspeed=6  (6")    ERROR* "8" works
Choice: 7 6         shutterspeed=7  7"      OK
Choice: 8 5         shutterspeed=8  8"      ERROR*  "5" works
Choice: 9 4         shutterspeed=9  4"      OK
Choice: 10 3.2      shutterspeed=10 10"     ERROR*  "3.2" works
Choice: 11 2.5      shutterspeed=11 2.5"    OK
Choice: 12 2        shutterspeed=12 2"      OK
Choice: 13 1.6      shutterspeed=13 13"     ERROR* "1.6" works
Choice: 14 1.3.     shutterspeed=14 1.3"    OK 
Choice: 15 1        shutterspeed=15 15"     ERROR ("1" "works" though)
Choice: 16 0.8      shutterspeed=16 0.8     OK
Choice: 17 0.6      shutterspeed=17 0.6     OK
Choice: 18 0.5      shutterspeed=18 0.5     OK
Choice: 19 0.4      shutterspeed=19 0.4     OK
Choice: 20 0.3      shutterspeed=20 20"     ERROR* ("0.3" works)
Choice: 21 1/4      shutterspeed=21 1/4     OK
Choice: 22 1/5      shutterspeed=22 1/5     OK
Choice: 23 1/6      shutterspeed=23 1/6     OK
Choice: 24 1/8      shutterspeed=24 1/8     OK
Choice: 25 1/10     shutterspeed=25 (25")   ERROR* ("1/10" works though)
Choice: 26 1/13     shutterspeed=26 1/13    OK
Choice: 27 1/15     shutterspeed=27 1/15    OK
Choice: 28 1/20     shutterspeed=28 1/20    OK
Choice: 29 1/25     shutterspeed=29 1/25    OK
Choice: 30 1/30     shutterspeed=30 30"     ERROR* ("1/30" works)
Choice: 31 1/40     shutterspeed=31 1/40    OK
Choice: 32 1/50     shutterspeed=32 1/50    OK
Choice: 33 1/60     shutterspeed=33 1/60    OK
Choice: 34 1/80     shutterspeed=34 1/80    OK
Choice: 35 1/100    shutterspeed=35 1/100   OK
Choice: 36 1/125    shutterspeed=36 1/125   OK
Choice: 37 1/160    shutterspeed=37 1/160   OK 
Choice: 38 1/200    shutterspeed=38 1/200   OK
Choice: 39 1/250    shutterspeed=39 1/250   OK
Choice: 40 1/320    shutterspeed=40 1/320   OK
Choice: 41 1/400    shutterspeed=41 1/400   OK
Choice: 42 1/500    shutterspeed=42 1/500   OK
Choice: 43 1/640    shutterspeed=43 1/640   OK
Choice: 44 1/800    shutterspeed=44 1/800   OK
Choice: 45 1/1000   shutterspeed=45 1/1000  OK
Choice: 46 1/1250   OK
Choice: 47 1/1600   OK
Choice: 48 1/2000   OK
Choice: 49 1/2500   OK
Choice: 50 1/3200   OK
Choice: 51 1/4000   OK
Choice: 52 1/5000   OK
Choice: 53 1/6400   OK
Choice: 54 1/8000   OK
```

msmeissn, 2019-10-26:
this seems just confusion between the "index" into the list and the "value".

For this we added --set-config variants which indicate if the index or if the value is to be used:

gphoto2 --set-config-index shutterspeed=25
gphoto2 --set-config-value shutterspeed=25

Does this help?

rpg777, 2019-10-27:
Great, if that works consistently across all the indices! Is that applied to aperture as well?

rpg777, 2019-10-27:
Also, it would seem to me that if gphoto2 reports out these indices, that is the expected use case. Why not just ensure that the index (code) is being applied??

msmeissn, 2021-02-28:
the same method also works for aperture.
it is kind of an old historic artefact that we have this function which can confuzse both index / value ... so use either the -index or -value variant please

closed 2021-02-28 by msmeissn