gphoto/libgphoto2 #456: Tether Between Linux (Ubuntu 18.04 LTS) Computer Running gphoto2 and Canon 5DS R (1.1.1) Intermittent Camera Freeze (

vepicurean, 2019-11-09:
I can reproduce this "freezing" behaviour on multiple Canon 5DS Rs when using a gphoto2 tether function to download images from my Canon EOS 5DS Rs to my Linux  computer (Ubuntu 18.04 LTS) using a script I wrote. 

I do not put an SD card in the 5DS Rs. I capture and download straight to the computer hard drives over the gphoto2 USB3 tether. I use high performance USB3 cables.

The frustrating thing is the issue is intermittent. It will run just fine and download hundreds of images with no problem for 15 mins or more during my capture sessions which can vary from about 50 captures to 500 captures (depends on the capture session). I capture and download a RAW image (about 50MB to 60MB) every 4 to 7 seconds.

Then there are times where on just the first or second capture the camera freezes (and stops capturing, though the mirrors and shutter are triggered, but the digital part of the camera is completely frozen), or will run fine for a while and then on the nth capture (say 88) the camera will seem to "freeze". 

See the attached image with this message "Recording.... Remaining images: 15" notice on the screen of the frozen camera. 

This message does not happen every time the camera freezes when using the gphoto2 tether (this message does not come up every time the camera freezes).  It's very random, so it's hard to reproduce, but it happens too frequently to the point I'm looking for other alternatives to tether and download the captured photos on the EOS 5DS Rs as this is too unstable.  Canon support is of no use. All the ever say is "send the camera in", but this is not something having to do with a faulty camera (and I can reproduce it on multiple cameras as I mentioned). There's a design or configuration issue somewhere with the Canon EOS 5DS R, gphoto2, or both. The Canon EOS 5DS R is running firmware 1.1.1  

[image]

Has anyone seen this before when using gphoto2 to download images from a Canon EOS camera? 
 
Thanks for your help and time. Greatly appreciated. I need to find a fix for this ASAP.

msmeissn, 2019-11-09:
This "image remaining" happens if we fail to download the image from the camera SDRAM, here we missed around 15 images.

This might happen if we miss or fail to handle object added events.

Is gphoto2 running all the time?

Is this happening if you shoot in very quick succession?

msmeissn, 2019-11-09:
(potential workaround would be to capture via the sd card (even for immediate download), but this might wear it out / be a bit slower)

vepicurean, 2019-11-09:
Hi, thanks for the reply. Appreciated.

It's the strangest thing. I have multiple tethered capture cameras using this gphoto2 tethered capture script, but I only have issues with only one setup. If I switch the cameras, the cameras that were working just fine before, exhibit the same intermittent "freezing" behaviour within the problematic tether setup, so it's either deep design issues with Canon (which they will never admit to, nothing is ever wrong design wise with their cameras according to them, everything is a return, which won't help in this case. They had the nerve to say send all the cameras back as somehow there's an issue with all the cameras that their repair shop can fix. It's such a racket),  or the script is just not stable or one of the functions has a bug/flaw. But then why do I only see the issue on the one tether setup? 

I set up the gphoto2 tether on all the cameras and the messaging comes back for each tether that the tether is waiting for photos to be captured. I went through the code and I notice we use two gphoto2 commands to setup and configure up the tether 

gphoto2 --capture-tethered=2    (which I don't know why there's a 2 there, I just noticed it. What does the 2 mean? Wait a certain amount of time?  It's not clear in the gphoto2 documentation)

or

gphoto2 --capture-tethered

Then I use a separate Arduino script to trigger the cameras using the the N3 connector. About every 4 to 6 seconds on average. But that can vary to a much longer time, as I have a sensor that can control that delay between triggers if the cameras don't need to be triggered. The fastest time between captures averages about 6 seconds. I do use "push twice to shoot", so the mirror goes up on the first push (short of the N3 connector circuit) and then the shutter captures on the second push (second short of the N3 connector circuit). It's fairly simple. If I walk up to the camera while the tether is waiting for a capture event and push the shutter twice, the images will download to the hard drive just fine. 

As I mentioned, the other tethers with the automated trigger/capture script work just fine.  

It's this one tether setup that has intermittent capture (camera freeze) issues. 

I physically change the USB ports, same intermittent failure behaviour. I change the cameras, same behaviour. I change the USB3 cables, same behaviour. I'm using high performance USB3 cables. Very frustrating. I've seen it run for an entire afternoon with no issues. The next day it won't run for 2 minutes without failing. 

I'm not getting you how the SD card is going to help? Will the camera copy the images from the SDRAM to the SD first card and then the gphoto2 captured tether will download the image from the SD card? Will it do this automatically?

vepicurean, 2019-11-12:
I notice Unbunti 18.04 is running libghoto2 2.5.16 which is a couple years old. Often having older versions of code is a good thing for stability, unless there is a known issue. 

I notice in the newer realeases of libghoto2 there have been some fixes for various Canon EOS models. But also a few things for broken and has to be fixed with an even newer version. 

What version of libghoto2 is best for the Canon 5DS Rs.

I ran tests on two different cameras again today.

I notice the problematic setup works fine for long periods of time, as I mentioned, then suddenly on a capture the camera freezes. There is an "unknown button 22" that comes across the libgphoto2 message logs right after this capture that causes the freeze. I have to take all the cables connected to the camera off, and then manually/physically push the shutter button a.couple times to unfreeze the camera. Again, I can replicate this on multiple RDS Rs, so it is not a camera specific issue. There is something external causing the camera to freeze. Not sure is that unknown button 22 message is cause of the freeze, or simply an effect of the freeze. Soon after the unknown button 22 message, gphoto2 reports an error across the tether log.

vepicurean, 2019-11-13:
More testing today. The camera basically became unusable with more attempted use through the gphoto2 tether throughout the day. Take the camera off gphoto2 tether, and it works fine. Put it back on the gphoto 2 tether and it can not get through more than three or four captures without freezing. At the start of the day, the camera on the gphoto2 tether did near 150 captures with no issues. This is consistent with what the others have been telling me they have been seeing on this one setup.  

Here's the gphoto2 messaging sequence with the camera freezes on the second push (shutter capture). 

[image]

then 

this messages shows on the gphoto2 tether message sequence and the camera is completely frozen. 

[image]

vepicurean, 2019-11-13:
Here's the message sequence when the capture is successful. 

[image]

vepicurean, 2019-11-13:
The problematic camera setup basically became so bad after a while, we had to stop shooting with the gphoto2 tether. We could get more than a couple captures successful over the tether. It's the strangest thing. The camera is not over heating, but the setup becomes completely unstable over time.

vepicurean, 2019-11-13:
I'm going to be doing some more testing on the tether with all the good and problematic cameras today. I have some additional ideas about what the issue is after doing some more research on these Canon 5DS Rs and various lenses. 

The one thing I'm worried about. The N3 connector is fairly "dumb". Basically you issue a short over the cable connected to the N3 connector. The camera is reacting to that short as you can see in the libgphoto2 event messages. Even when the freeze happens, the camera is getting the second short via the N3 connector, and it's during this capture that the camera freezes. 

I am just wondering if the is some "conflict" command or message that is coming from gphoto2 that is confusing the problematic camera and causing the freeze?

vepicurean, 2019-11-17:
I think I'm onto the issue. I can control (eliminate) the freezing by using different prime lenses (different focal lengths).  It's the strangest thing. 

I'm doing more tests with various prime lenses with libgphoto2 connected vs not connected. 

I'm quite disappointed in the performance of these very expensive Canon DLSR camera bodies and these expensive prime lenses. 

I'm just running some more experiments to determine if libgphoto2 is involved with the environment that induces the freezing or not. Will keep you updated.

vepicurean, 2021-03-02:
I know it has been a long time. There's something going on between the Sigma 50mm prime lens and the Canon 5DS R camera. Neither Canon, and to my surprise Sigma want to discuss the issue. Canon hid behind the "we only support issues when our lens is used", which makes me thing they are doing something to purposely mess things up. But the strange thing is the Sigma 35mm prime lens does not have the freezing issue. When I proved this to Sigma that for some reason their 50mm lens seems to freeze the 5DS R camera body, they didn't want to discuss it. Saying their have not seen it themselves (which is not really true, they have had support tickets I notice for freezing issues on these lenses and firmware fixes for similar issues), 

So in summary, you can't use a Sigma 50mm prime Art series lens for rapid tethered capture using gphoto on a Canon 5DS R camera body. But the 35MM prime lens from that Art series works fine for rapid tethered capture using gphoto on a Canon 5DS R body. You have no idea how many hours it took me to figure this out. And both companies ran and hit when I showed them the video evidence. 

Using the Sigma 35mm prime Art series is just a work-around. Really for these type of captures, we should be using the 50mm focal length, but the camera go crazy almost immediately with tethered and using gphoto. Having that huge 50mm Sigma lens facing down (lens vertical, and the camera body bearing the weight of the lens) may also be an issue. I mentioned to Sigma and they didn't want to hear it. 

Just a heads up for others that may run into this issue while doing tethered capture using Gphoto with a Sigma 50mm Art series length on a Canon EOS camera.