gphoto/libgphoto2 #461: --capture-image taking 40 seconds on Nikon Coolpix S2600

AdvancedStyle, 2019-11-28:
**Describe the bug**

When doing capture-image on a Nikon Coolpix S2600 the command takes a photo right away (within 2 seconds), but then locks up the camera for a long time (takes 35 seconds to return), and until the 35 seconds is over I cannot do anything (such as download the images or get the file list)

**Name the camera**
`Nikon Coolpix S2600 (PTP mode) usb:001,045`

**libgphoto2 and gphoto2 version**
```
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

**To Reproduce**
gphoto2 --capture-image --debug --debug-logfile=debug.log --debug-loglevel debug

[debug.log](https://github.com/gphoto/libgphoto2/files/3902168/debug.log)

AdvancedStyle, 2019-11-30:
I've created a pull request to fix this issue: https://github.com/gphoto/libgphoto2/pull/462

msmeissn, 2019-11-30:
i merged the pull request now :) 
I assume it works better than the flaky event handlign

closed 2019-11-30 by msmeissn