gphoto/libgphoto2 #464: No capture support on Canon G9X Mark II

guischmitd, 2019-12-12:
**Describe the bug**

When I plug a powershot G9X Mk II via USB to my Ubuntu 18.04 64 bit machine, the camera switches to the image viewer mode. [This user](https://github.com/gphoto/libgphoto2/issues/379#issuecomment-469097788) states they're able to capture with the same camera.

gphoto2 --summary outputs:
```
Camera summary:                                                                
Manufacturer: Canon Inc.
Model: Canon PowerShot G9 X Mark II
  Version: 1-15.0.1.0
  Serial Number: DD93FE0F2949A441243132872F5188D7
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS Wave, JPEG, Defined Type, Unknown(b103), Unknown(b982), Unknown(b105), Unknown(bf01)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:
store_00010001:
	StorageDescription: SD
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 31902433280 (30424 MB)
	Free Space (Bytes): 25465716736 (24286 MB)
	Free Space (Images): -1

Device Property Summary:
Event Emulate Mode(0xd045):(readwrite) (type=0x4) Enumeration [1,2,3,4,5,6,7] value: 2
Property 0xd04a:(readwrite) (type=0x2) Enumeration [0,1,2,3] value: 0
Size of Output Data from Camera(0xd02e):(read only) (type=0x6) 524288
Size of Input Data to Camera(0xd02f):(read only) (type=0x6) 524288
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3] value: 3% (3)
Battery Type(0xd002):(read only) (type=0x4) Enumeration [0,1,2,3,4,5] value: Unknown (0)
Battery Mode(0xd003):(read only) (type=0x6) Enumeration [0,1,2,3] value: Normal (1)
UNIX Time(0xd034):(readwrite) (type=0x6) 1576161853
Type of Slideshow(0xd047):(read only) (type=0x4) 0
DPOF Version(0xd046):(read only) (type=0x4) 257
Remote API Version(0xd030):(read only) (type=0x6) 256
Model ID(0xd049):(read only) (type=0x6) 68157440
Camera Model(0xd032):(read only) (type=0xffff) 'Canon PowerShot G9 X Mark II'
Camera Owner(0xd033):(readwrite) (type=0x4002) a[0] 
Firmware Version(0xd031):(read only) (type=0x6) 16777216
Property 0xd050:(read only) (type=0x2) 0
Property 0xd051:(read only) (type=0x0) Undefined
Property 0xd052:(read only) (type=0x2) 0
Property 0xd053:(readwrite) (type=0x4002) a[0] 
Property 0xd054:(readwrite) (type=0x4002) a[0] 
Property 0xd057:(read only) (type=0xffff) '852053000854'
Property 0xd402:(read only) (type=0xffff) 'Canon PowerShot G9 X Mark II'
Property 0xd406:(readwrite) (type=0xffff) 'Windows'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1

```

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
```
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:001,028    
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
```
gphoto2 2.5.15

Copyright (c) 2000-2017 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

```
**To Reproduce**
Simply connect a powershot G9X Mark II to a Linux machine Ubuntu

hfiguiere, 2019-12-12:
This version of libgphoto2 is older than the camera.

guischmitd, 2019-12-13:
Thank you very much, I'll upgrade and try again!

closed 2019-12-13 by guischmitd