gphoto/libgphoto2 #468: Cannot set exposure compensation on Sony RX0 II

romainreignier, 2020-01-22:
As explained in https://github.com/gphoto/libgphoto2/issues/419#issuecomment-508028121 the RX0 II seems to work correctly with libgphoto2 unless when I try to set the exposure compensation setting.

Getting the value is ok :
```
$ gphoto2 --get-config=exposurecompensation
Label: Exposure Compensation                                                   
Readonly: 0
Type: RADIO
Current: -0.7
END
```

But setting any value will cause a change but only to the value 0:
```
$ gphoto2 --set-config=exposurecompensation=-1.0
```

In my previous comment, I was using the firmware 1.0 and I have now upgraded it to the latest 2.0 firmware but the issue is still present.

I have attached a log file of the operation on a Sony RX0 II and the same command on a RX0 I on which it works.
[set_exposurecompensation_rx0I.log](https://github.com/gphoto/libgphoto2/files/4097783/set_exposurecompensation_rx0I.log)
[set_exposurecompensation_rx0II.log](https://github.com/gphoto/libgphoto2/files/4097784/set_exposurecompensation_rx0II.log)

Thank you

romainreignier, 2020-01-22:
While comparing the two log files, I have found these lines which seems suspect:
On the RX0 II:
```
0.104869 gp_widget_set_value         (2): Setting value of widget 'Exposure Compensation' to '-1.0'...
0.104972 _set_config                 (2): Setting property 'Exposure Compensation' / 0x5010
0.104996 _put_sony_value_i16         (2): setting 0x5010 to 0x00000000
```
While on the RX0 I:
```
0.092968 gp_widget_set_value         (2): Setting value of widget 'Exposure Compensation' to '-1.0'...
0.093036 _set_config                 (2): Setting property 'Exposure Compensation' / 0x5010
0.093060 _put_sony_value_i16         (2): setting 0x5010 to 0xfffffc18
```
So it looks like libgphoto sends `0x00000000` instead of `to 0xfffffc18`.

romainreignier, 2020-01-22:
From the code of [_put_ExpCompensation](https://github.com/gphoto/libgphoto2/blob/e14e6154715ae4673ab2bd0332cc55f1d597c835/camlibs/ptp2/config.c#L1462-L1483), the value is 0 because `dpd->FORM.Enum.NumberOfValues` is empty for the RX0 II.

Indeed, the command `gphoto2 --get-config=exposurecompensation` returns:
```
Label: Exposure Compensation                                                   
Readonly: 0
Type: RADIO
Current: 0
END
```

While on RX0 I:
```
Label: Exposure Compensation                                                   
Readonly: 0
Type: RADIO
Current: -1
Choice: 0 0
Choice: 1 0
Choice: 2 5
Choice: 3 4.7
Choice: 4 4.5
Choice: 5 4.3
Choice: 6 4
Choice: 7 3.7
Choice: 8 3.5
Choice: 9 3.3
Choice: 10 3
Choice: 11 2.7
Choice: 12 2.5
Choice: 13 2.3
Choice: 14 2
Choice: 15 1.7
Choice: 16 1.5
Choice: 17 1.3
Choice: 18 1
Choice: 19 0.7
Choice: 20 0.5
Choice: 21 0.3
Choice: 22 0
Choice: 23 -0.3
Choice: 24 -0.5
Choice: 25 -0.7
Choice: 26 -1
Choice: 27 -1.3
Choice: 28 -1.5
Choice: 29 -1.7
Choice: 30 -2
Choice: 31 -2.3
Choice: 32 -2.5
Choice: 33 -2.7
Choice: 34 -3
Choice: 35 -3.3
Choice: 36 -3.5
Choice: 37 -3.7
Choice: 38 -4
Choice: 39 -4.3
Choice: 40 -4.5
Choice: 41 -4.7
Choice: 42 -5
END
```

But I am not familiar enough with libgphoto2 and ptp to investigate more on why these possible values are not retrieved correctly.

qtnemo, 2020-01-22:
We have what looks like an exact problem with out Sony Alpha 7R4.  The ISO value can be read, but not set.  On a 7R3, --get-config lists all the choices and set-config works.  But on a 7R4, the type is still RADIO, but there are no choices, and --set-config won't change the setting.  I think we posted an ticket here with more details but now I could not find my post.  I believe the error message return from libgphoto2 to our C/C++ call to gp_widget_set_value (child, X)  is "X is not one of the enumerated value".  We tried obvious choices like 0, 1, 2, ... and actual ISO values like 100, 800, ... but the error is the same and the ISO value remains unchanged.

romainreignier, 2020-01-22:
Interesting @qtnemo 
Sony must have changed something in their PTP protocol.

TimRyan99, 2020-08-30:
I have tried setting exposure compensation in Sony RX0 M2 (v2.0 firmware) and ISO in gphoto2 but without success. I see in gphoto2 --config that the ISO and exposure compensation fields are present but greyed out and are not accessible. Also Entangle does not allow these parameters to be changed either. I understand Entangle uses libphoto2 library so presume the issue is to do with this library and not gphoto2 or Entangle. Finally I have checked on Windows machine with Sony ImageEdge-Remote and find I am able to make these changes. So looks like this is not a limitation of the camera. Appreciated any assistance. Seems like this issue might be cropping up in other Sony cameras.

msmeissn, 2020-08-31:
I have adjustments for this in current libgphoto2 GIT, will be in 2.5.26 release.

romainreignier, 2020-09-07:
@msmeissn I have tested the latest git version on RX0 II and the problem is solved, I can set the exposure compensation correctly and faster now.
Thank you!

msmeissn, 2020-09-12:
thx for the good feedback! :)

closed 2020-09-12 by msmeissn