gphoto/libgphoto2 #469: Olympus E-M1 Mark II PTP General Error

Kelvin-Wu, 2020-01-24:
Hi All,
I use gphoto/libgphoto2 2.5.23 on ubuntu 19.10.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

I can use --config to setting camera.
But when I use gphoto2 --capture-image   
It show:

ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***      
PTP General Error

What should I do to fix this isue?

msmeissn, 2020-01-26:
can you attach a debuglogfile?

gphoto2 --debug --debug=logfile.txt --capture-image

Kelvin-Wu, 2020-01-26:
Hi msmessn,
Sorry I'm a new linux user. I've done it, but where is the logfile.txt?

Kelvin-Wu, 2020-01-26:
Hi @msmeissn ,
I used save function to saved a log file.
At the end. The Terminal repeat as below shown, so I have to turn off my camera to stop it.

10.054973 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
10.255115 gp_port_get_timeout         (2): Current port timeout is 20000 milliseconds.
10.255138 gp_port_set_timeout         (2): Setting port timeout to 150 milliseconds.
10.255142 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
10.405386 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
10.555655 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
10.755827 gp_port_get_timeout         (2): Current port timeout is 20000 milliseconds.
10.755893 gp_port_set_timeout         (2): Setting port timeout to 150 milliseconds.
10.755922 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
10.906199 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...

[gphoto2_log.txt](https://github.com/gphoto/libgphoto2/files/4113666/gphoto2_log.txt)

msmeissn, 2021-10-03:
we made some bugfixes in newer libgphoto2 2.5.27 for the olympus capture... can you try with a newer release?